kiibohd/controller #5: Ubuntu LTS 14.04 Serial Terminal Blank

haata, 2015-02-09:
Ubuntu LTS 14.04 has an issue with the Virtual Serial Port

screen /dev/ttyACM0

Terminal will accept commands, but will not show any text. It is likely an issue with either the kernel or some external program Ubuntu uses. This is likely fixable through changes in the virtual serial port descriptor, but it's not obvious what needs to be fixed.

Not an issue on Windows, Mac, or Arch Linux.

haata, 2015-02-09:
Typing led then enter will turn the led on.

haata, 2015-02-20:
Same issue confirmed for Lunar Linux kernel 3.10.0.

blais, 2015-03-07:
Ditto for Ubuntu 13.04 (on a MacbookPro).

smasher816, 2015-03-09:
I know others have this issue; however, it does not seem to happen to me when I made a Ubuntu 14.04 LTS guest on my macbook. Perhaps I will try out 13.04 or some older release.

http://i.imgur.com/GJgM9ih.png

haata, 2015-03-09:
Interesting. I really thing it is kernel related (I used to know the version exactly). Basically there was a big change in the cdc_acm kernel module sometime between 3.10 and 3.16

smasher816, 2015-03-09:
I guess LTS recently had some updates the included the 3.16 kernel? That seems to be what my uname is saying. I grabbed the 14.04 iso from ubuntu's website last night and did a fresh install...

So I suppose you could put that in the readme for some people to consider doing (running the updates if your on the LTS). Kernel upgrades would most likely also work for other distributions.

If we really want to dig into what happened with the kernel module I'll grab an older version of the kernel or an older OS and see what I can figure out after I finish up my work.

haata, 2015-03-09:
This problem may just end up going away, but I'm not sure entirely. Even if we don't fix it, I sorta want to have enough information to understand why it's not working.

blais, 2015-03-09:
Thanks guys.
I'm due for a backup + upgrade cycle anyway; will do and report.

smasher816, 2015-03-09:
I played around a little bit more with ubuntu. If I install the 3.10 kernel then the issue is present, and if I boot the 3.16 one then it is gone. Therefore this is definitely a kernel issue, not a software one. For the time being it might be wise to add a little disclaimer saying you must be using kernel 3.XX or above for the serial console to work.

For those that do not want to do a full distro upgrade it is probably possible to simply add the new kernel. 1) Go to this website, http://kernel.ubuntu.com/~kernel-ppa/mainline, and open a folder you want, ex: v3.19.1-vivid.
2) You must then download three debs: The _all one, and then depending on your archictecture (32/64bit) the _amd64 or _i386 header and image.
3) Put all these three debs in one folder, and open a terminal in that folder.
4) Run "sudo dpkg -i *.deb" to install the kernel packages, and "sudo update-grub" to update the boot menu.
5) While booting you can hold shift and the grub menu will appear, go to the advanced section and you can choose which kernel version you want to boot.

If this works then you can later remove the older kernels with "sudo apt-get purge linux-image-x.x.x-x-generic", which I will believe make the newer one the default. You can confirm what version you are running with "uname -r".

smasher816, 2015-03-10:
The converse is also true for debugging. One can install an old 3.10 kernel and experience the issue, and thus test.

Haata: Do you happen to know the exact version that fixed the issue? It would help with narrowing down the changes. Fwiw, here is a diff between the 3.10 module and the 3.16 module. http://ix.io/gP2

blais, 2015-03-19:
FYI.
I updated to utopic kernel:

Linux <hostname> 3.16.0-31-generic #43~14.04.1-Ubuntu SMP Tue Mar 10 20:13:38 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

and I'm now able to debug the keyboard.

tmk, 2015-07-09:
I've had this problem on Ubuntu 14.04 and 14.10 but no problem  for unclear reason now.
Not confirmed completely but I highly suspect **modemmanager** as culprit of this.

The otherday I had similar problem on Arduino bootloader(caterina)'s CDC serial port and resolved it after I stopped and deinstalled **modemmanager** daemon.

haata, 2015-08-19:
I've cleaned up the virtual serial port section of my USB stack. This may help with some of the issues.

20b62afb9a05a64b1f15c6329866600913775eea

haata, 2019-02-18:
I don't believe this should be an issue any longer.

closed 2019-02-18 by haata