kiibohd/controller #6: Add program detection to load script

haata, 2015-02-10:
Add program detection to load scripts. LoadFile/*

* Make sure mac support is working
* Find command-line serial utility for Cygwin/Windows (screen doesn't work on Windows for serial ports)
* Check for all needed programs before running them

smasher816, 2015-02-13:
I submitted a simple pull request to fix the third bullet point. Regarding the other two points:

I have a mac and can compile on it fine. As far as I remember the instructions were fine and I just had to install the build tools and some other required programs like dfu-util.

I'm not sure of many command line serial programs for windows. Some people say that screen works under cygwin... A quick google says that powershell can be used to read and right lines but it is not a full fledged screen replacement. If graphical programs are ok I think Putty is the best bet - it is a connection program with a compatible terminal emulator for ssh, telnet, serial, etc. See here: http://www.chipkin.com/using-putty-for-serial-com-connections-hyperterminal-replacement/. If that is fine I can update the readme and this can be closed.

haata, 2019-02-18:
With the upcoming HID-IO daemon, nothing should be required outside of the load scripts that are already written.

closed 2019-02-18 by haata