kiibohd/controller #9: Will this work with Atmega32a?

aaronz8, 2015-02-17:
I'm attempting to find a replacement firmware for my FaceW pcb (to replace ps2avru). Problem is, it requires software USB support and I'm not sure if kiibohd has that built in.

Thanks!

haata, 2015-02-18:
Yeah, I don't really have support for software usb. Though It can be added (I think tmk has experimental virtual usb support).

The other big problem is the amount of RAM and Flash on the atmega32a.
32k of Flash is not that much and 2k of SRAM is definitely not enough right now.

The macro'ing system is basically an unrolled regex, which requires quite a bit of state and all of those state transitions need to be stored in flash.
I am planning on implementing a low RAM usage version of the macro (kll) engine that will only support macro combinations (no sequences) and layers. That might have a chance at fitting.

I've been transitioning away from AVR over the past few years because it's just more annoying to work with than ARM (dealing with flash is soooo much easier). avr-gcc also tends to fluctuate a lot.

Anyways, I'm happy to help out a bit if you want to submit some patches, but otherwise, good luck in finding something :D

aaronz8, 2015-02-18:
Thanks for the explanation! Very helpful. I think at this point I might just design my own pcb rather than trying to get this FaceW to work, haha.

closed 2015-02-18 by aaronz8

haata, 2015-02-18:
Haha, g'luck.

The infinity pcb was open-sourced very recently. Feel free to use it. The tmk has also been ported to it.