kiibohd/controller #11: Numkeypad in windows vs linux

terrypchan, 2015-02-20:
I have a numkeypad defined on layer2, which works fine in windows, but same keyboard under linux does not work. The switch to layer2 works but the numkeypad types garbage on the screen. Attached is a dropbox link to a zip file containing the kll and *.h and .bin file. Thanks.
 
https://www.dropbox.com/s/nk9qfh0ya0m5i9w/tchan20150220.zip?dl=0

haata, 2015-02-21:
Replied fix on irc

KEYPAD_DECIMAL and KEYPAD_PERIOD are two different keys, the common one for keyboards is KEYPAD_PERIOD. KEYPAD_DECIMAL is hex vs. decimal.

Just use U"Keypad Period" or U"P."

I have noticed some strange-ness with KEYPAD_DECIMAL though, keeping this issue open until I verify.

haata, 2015-02-24:
I've verified (using the new outputDebug cli option) that the keyboard is sending the USB codes correctly. Linux was just dealing with the key strangely (as expected because it's non-standard).

closed 2015-02-24 by haata