kiibohd/controller #18: Unable to build macro using shift key

mcmasterathl, 2015-02-27:
Putting this in a kll file does not produce expected results:
```
U"Shift" + U"Esc" : U"Backtick";
```

But this does:
```
U"Ctrl" + U"Esc" : U"Backtick";
```

Relevant snippet from CMakeLists.txt
```
set(     BaseMap "defaultMap"
	CACHE STRING "KLL BaseMap/Scancode Keymapping" )

##| Layer additonal .kll maps on the BaseMap, layers are in order from 1st to nth
##| Can be set to ""
set(  DefaultMap "mcmLayer0 stdFuncMap"
	CACHE STRING "KLL DefaultMap" )

##| ParitalMaps available on top of the BaseMap. See above for syntax on specifying multiple layers vs. layering
##| Can be set to ""
set( PartialMaps "mcmLayer1"
	CACHE STRING "KLL PartialMaps/Layer Definitions" )
```

mcmLayer0.kll:
```
Name = mcmLayer0;
Version = 0.1;
Author = "Steve McMaster";
KLL = 0.3;
Date = 2015-02-20;
U"Function2" : U"Function1";
U"Backtick" : U"Delete";
```

mcmLayer1.kll:
```
Name = mcmLayer1;
Version = 0.1;
Author = "Steve McMaster";
KLL = 0.3;
Date = 2015-02-20;
U"Esc" : U"Backtick";
U"1" : U"F1";
U"2" : U"F2";
U"3" : U"F3";
U"4" : U"F4";
U"5" : U"F5";
U"6" : U"F6";
U"7" : U"F7";
U"8" : U"F8";
U"9" : U"F9";
U"0" : U"F10";
U"-" : U"Home";
U"=" : U"End";
U"Backslash" : U"PageUp";
U"Backtick" : U"PageDown";
U"Backspace" : U"Delete";
U"W" : U"Up";
U"A" : U"Left";
U"S" : U"Down";
U"D" : U"Right";
U"{" : U"Up";
U";" : U"Left";
U"'" : U"Right";
U"/" : U"Down";
```

Instead of getting a backtick when pressing Shift + Esc, escape is produced.

mcmasterathl, 2015-02-27:
Also, generatedKeymap.h because either GitHub won't allow me to attach a file or I can't figure out how:

```
/* Copyright (C) 2014-2015 by Jacob Alexander
 *
 * This file is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This file is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this file.  If not, see <http://www.gnu.org/licenses/>.
 */

// This file was generated by the kll compiler, DO NOT EDIT.
// Generation Date:    2015-02-27
// KLL Backend:        kiibohd
// KLL Git Rev:        867e7495a5cc050ef5c93fabd2a66eadf170e60d
// KLL Git Changes:    None
// Compiler arguments:
//    ../kll/kll.py
//      /Users/steve/git/github/controller/Macro/PartialMap/capabilities.kll
//      /Users/steve/git/github/controller/Output/pjrcUSB/capabilities.kll
//      /Users/steve/git/github/controller/Scan/MD1/defaultMap.kll
//      -d
//      /Users/steve/git/github/controller/kll/layouts/mcmLayer0.kll
//      /Users/steve/git/github/controller/kll/layouts/stdFuncMap.kll
//      -p
//      /Users/steve/git/github/controller/kll/layouts/mcmLayer1.kll
//    --backend
//      kiibohd
//    --templates
//      /Users/steve/git/github/controller/kll/templates/kiibohdKeymap.h
//      /Users/steve/git/github/controller/kll/templates/kiibohdDefs.h
//    --outputs
//      generatedKeymap.h
//      kll_defs.h
//
// - Base Layer -
//    PartialMapCapabilities
//      /Users/steve/git/github/controller/Macro/PartialMap/capabilities.kll
//    pjrcUSBCapabilities
//      /Users/steve/git/github/controller/Output/pjrcUSB/capabilities.kll
//    MD1
//      /Users/steve/git/github/controller/Scan/MD1/defaultMap.kll
// - Default Layer -
//    mcmLayer0
//      /Users/steve/git/github/controller/kll/layouts/mcmLayer0.kll
//    stdFuncMap
//      /Users/steve/git/github/controller/kll/layouts/stdFuncMap.kll
// - Partial Layers -
//    Layer 1
//     mcmLayer1
//       /Users/steve/git/github/controller/kll/layouts/mcmLayer1.kll

#ifndef __kiibohdKeymap_h
#define __kiibohdKeymap_h

// ----- Includes -----

// KLL Include
#include <kll.h>

// ----- Capabilities -----

[... 529 lines cut ...]
default_tl_0x00, default_tl_0x01, default_tl_0x02, default_tl_0x03, default_tl_0x04, default_tl_0x05, default_tl_0x06, default_tl_0x07, default_tl_0x08, default_tl_0x09, default_tl_0x0A, default_tl_0x0B, default_tl_0x0C, default_tl_0x0D, default_tl_0x0E, default_tl_0x0F, default_tl_0x10, default_tl_0x11, default_tl_0x12, default_tl_0x13, default_tl_0x14, default_tl_0x15, default_tl_0x16, default_tl_0x17, default_tl_0x18, default_tl_0x19, default_tl_0x1A, default_tl_0x1B, default_tl_0x1C, default_tl_0x1D, default_tl_0x1E, default_tl_0x1F, default_tl_0x20, default_tl_0x21, default_tl_0x22, default_tl_0x23, default_tl_0x24, default_tl_0x25, default_tl_0x26, default_tl_0x27, default_tl_0x28, default_tl_0x29, default_tl_0x2A, default_tl_0x2B, default_tl_0x2C, default_tl_0x2D, default_tl_0x2E, default_tl_0x2F, default_tl_0x30, default_tl_0x31, default_tl_0x32, default_tl_0x33, default_tl_0x34, default_tl_0x35, default_tl_0x36, default_tl_0x37, default_tl_0x38, default_tl_0x39, default_tl_0x3A, default_tl_0x3B, default_tl_0x3C, default_tl_0x3D, default_tl_0x3E
};

// - Partial Layer ScanCode Lookup Maps -
// Partial Layer 1
const nat_ptr_t *layer1_scanMap[] = {
layer1_tl_0x00, layer1_tl_0x01, layer1_tl_0x02, layer1_tl_0x03, layer1_tl_0x04, layer1_tl_0x05, layer1_tl_0x06, layer1_tl_0x07, layer1_tl_0x08, layer1_tl_0x09, layer1_tl_0x0A, layer1_tl_0x0B, layer1_tl_0x0C, layer1_tl_0x0D, layer1_tl_0x0E, layer1_tl_0x0F, layer1_tl_0x10, layer1_tl_0x11, layer1_tl_0x12, layer1_tl_0x13, layer1_tl_0x14, layer1_tl_0x15, layer1_tl_0x16, layer1_tl_0x17, layer1_tl_0x18, layer1_tl_0x19, layer1_tl_0x1A, layer1_tl_0x1B, layer1_tl_0x1C, layer1_tl_0x1D, layer1_tl_0x1E, layer1_tl_0x1F, layer1_tl_0x20, layer1_tl_0x21, layer1_tl_0x22, layer1_tl_0x23, layer1_tl_0x24, layer1_tl_0x25, layer1_tl_0x26, layer1_tl_0x27, layer1_tl_0x28, layer1_tl_0x29, layer1_tl_0x2A, layer1_tl_0x2B, layer1_tl_0x2C, layer1_tl_0x2D, layer1_tl_0x2E, layer1_tl_0x2F, layer1_tl_0x30, layer1_tl_0x31, layer1_tl_0x32, layer1_tl_0x33, layer1_tl_0x34
};

// ----- Layer Index -----

// -- Layer Index List
//
// Index 0: Default map
// Index n: Additional layers
const Layer LayerIndex[] = {
	Layer_IN( default_scanMap, "D: mcmLayer0 + stdFuncMap", 0x00 ),
	Layer_IN( layer1_scanMap, "1: mcmLayer1", 0x00 ),
};

// - Layer State
uint8_t LayerState[ LayerNum ];

#endif // __generatedKeymap_h
```

mcmasterathl, 2015-02-27:
Actually, I realized that I had mapped the key i THOUGHT was control to function 1, and also conveniently made myself function 1 + escape is backtick. So, the control option doesn't work either.

mcmasterathl, 2015-02-28:
As best as I can tell with my limited understanding of how the KLL compiler works, it looks like in the `eval_usbCode` function (kll.py, line 427ish), the triggers has the properly expanded KEY_LCTRL and KEY_BACKTICK, but when it calls macros_map.lookupUSBCodes, the combination of those two isn't found anywhere. I'm not sure how to go about investigating this further, but I'm happy to help if you can point me in the right direction.

haata, 2015-02-28:
Yeah, kll.py and kll_lib/containers.py are responsible for the kll file parsing and correlation.

backends/kiibohd.py is responsible for generating the .h file output. I'd probably start here to see if the correct data is available in the data structures.
Unforunately, these data structures are multi-dimensional lists so it gets a bit confusing to think about (I'm going to have to add more dimensions for the next draft of kll too...).
The easiest way to think about it though: Sequence of combinations. The inner most list deals with key combinations, where level up you are looking at sequences. It starts to get tricky when you get into dealing with the range definitions of kll (so hopefully you don't have to dig that deep).

mcmasterathl, 2015-02-28:
So this works:
```
S0x1D + S0x00 : '1';
```

Which in my case is Control + Escape. If I use S0x2A instead of S0x1D (LShift), then I get an "!" *and* an escape character (that is, if I open vim, go into insert mode and press shift + esc, I get a "!" and it exits insert mode, as if I had typed "!" and then pressed escape.

Hopefully I'm getting closer?

haata, 2015-02-28:
So, that sorta makes sense with the current implementation.

First, it sounds like scan codes are definitely working and the issues has to do with the USB -> Scan translation (the firmware uses Scan codes internally as that's the most efficient lookup).

Second, ! is expected, because the keyboard is also sending the shift key at the same time. Right now there isn't a way to block shift key from sending shift. Is this a feature you want?
It would involve changes to the KLL and compiler so it would a bit lower on my priority list right now (bug fixing first :P).

mcmasterathl, 2015-02-28:
I think it would be neat but since my end goal is shift + esc = ~, which itself requires shift, I don't think this is in particular a problem. I can get by with using Scan codes for now.

I'll keep poking at the KLL compiler because I'm having fun and see if I can make the USB -> Scan translation work for combinations and sequences (I tested U"Shift" , U"Esc" as well, also doesn't work).

haata, 2015-02-28:
It's a fun little compiler.
But it was a bit of a rush job so it doesn't have the error and debug reporting that I'd like it to have.

I'm also working on a kll -> json backend (as well as a json -> kll option) to better support web configurators. The massdrop configurator was only intended as a temporary thing until we could find someone to write us a full featured one (and support future features as well). All of it will be opensource.

mcmasterathl, 2015-02-28:
I submitted a PR to the KLL compiler that seems to fix the issue with not supporting combinations/sequences with USB codes as triggers. Works in my testing.

I will keep this issue around in case you decide to add support for blocking the Shift key in combinations.

haata, 2015-02-28:
Thanks a lot! The patch looks good, but I want to run it through a couple of tests first. I should be able to properly review it this weekend.

Shift key blocking will take changes in a bunch of places as the state of the shift key must be tracked by the macro using the shift key.

mcmasterathl, 2015-02-28:
Understood. Not really a high priority feature for me at this point, just figured in case you get bored down the road I'll leave the issue laying around.

haata, 2015-08-19:
While not the full feature, I did implement an example of how to achieve this using capabiltiies.
It's somewhat hack'ish but seems to work ok (at least until I can get KLL straightened out for doing it properly).

joetoth, 2015-11-10:
I'm having a similar problem. I would like to map Space+Q = 1.

Infinity keyboard using infinity.bash with my custommap
DefaultMap="md1Overlay mycustommap stdFuncMap"
that contains:

U["Space"] : blockHold( 0x10 );
U"Q" : blockKey( 0x10 );
U["Space"] + U"Q" : '1';

Space+Q does work, but Q by itself now emits 'm' and space does not work.

Thoughts on what I could try?

LogicalDash, 2016-05-07:
I tried the blockHold hack on my Ergodox Infinity, like so:
```
blockHold => CustomAction_blockHold_capability( usbCode : 1 );
U"1" : blockHold(0xe1);
U"1" :+ U"Shift" + U"7";
U"Shift" + U"1": blockHold(0xe1);
U"Shift" + U"1":+ U"Shift" + U"5";
```
(I pasted the examples added in 398018e into MDergo1/scan_loop.c)

Pressing 1 gets me the ampersand, but Shift+1 *also* gets me the ampersand, as if the shifted combinations didn't take. Same story if I substitute blockKey.

burnedmonk, 2016-07-22:
hi,
edit the "scancode_map.kll" is the best way to make dvorak programmer layout(on infinity keyboard)?

S0x01 : U"7"+U"LShift"; 
S0x01 + S0x2A : U"5"+U"LShift";
S0x01 + S0x35 : U"5"+U"LShift";
work! 

i try same thing in base layer
S0x01 : '&'; 
S0x2A + S0x01  : '%'; 
S0x35 + S0x01 : '%';
if I press the S0x01 the keyboard print '&' 
if I press S0x01 and (S0x2A or S0x35) the keyboard print "%&"

http://www.kaufmann.no/roland/dvorak/images/dvp1.png

haata, 2016-07-22:
The easiest way is to use a kll "overlay".

I have a Colemak example here -> https://github.com/kiibohd/kll/blob/master/layouts/colemak.kll

burnedmonk, 2016-07-22:
I put this :
'1':'&'
'!':'%'
In infinity60%.bash ad default map...

If I press 1+Lshift to print '%' the keyboard print "%&"

If I press 1+Rshift to print '%' the keyboard print "&"

haata, 2016-07-23:
Ah, yes. So, you have two options for that.

1) Wait till KLL 0.5 (still rewriting the compiler for support of this)
2) Write a custom capability in C, that handles key blocking

burnedmonk, 2016-07-25:
Thanks for the reply! 
I'll wait the .5 ;)