kiibohd/controller #19: Sequence trigger behavior

mcmasterathl, 2015-02-28:
So if I use either of these in my KLL file:

```
S0x1D , S0x2B : U"5";
U"Control" , U"z" : U"5";
```

Then after plugging the keyboard in, pressing control and then z results in a z *then* a 5, occasionally with the 5 repeating itself until another key is pressed. After this, the sequence no longer works. The same behavior is experienced with both the Scan codes and USB codes.

haata, 2015-02-28:
Interesting, looks like the state is not handled correctly.

The keyboard itself has a debug terminal (doesn't work in Ubuntu 14.04 LTS btw).
It's a virtual serial port (on Linux, screen /dev/ttyACM0, Mac it's something about usb in the /dev directory).

It's not the most user friendly thing, but there are commands to query each macro at it's current state:
macroList
macroShow T25
to show the state of Trigger macro 25

The easiest way to track down the trigger macro number is by looking at the generatedKeymap.h file.

I have an idea what's wrong, but I'll have to dig a bunch.
Thanks a lot for hard work :D

haata, 2015-03-02:
Bug has been confirmed, now working on a fix.

kaansoral, 2015-05-09:
I also tested this, the behaviour is very non-deterministic/erratic, I'm assuming the bug still exists?

The sequence trigger behaviour is extremely useful, it think it should also momentarily drop modifiers and enter the sequence

For example, U"Ctrl",U"B":U"C" - should print "C" instead of sending Ctrl+Z

So it should reset modifiers, apply the sequence, reset the modifiers to it's previous state (probably keep a stack, and pop back)