kiibohd/controller #22: Fix UNKNOWN name in dfu bootloader

haata, 2015-03-02:
dfu-util --list 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Found DFU: [1c11:b007] ver=0000, devnum=19, cfg=1, intf=0, alt=0, name="UNKNOWN", serial="mk20dx128vlf5"

This is non-upgradeable code. But this should just be a cosmetic fix.

smasher816, 2015-03-08:
Here is some guidance:

Each device interface can have multiple alternative settings as described here - http://www.beyondlogic.org/usbnutshell/usb5.shtml#AlternateSetting. Section 4.2.3 of the USB DFU interface may be helpful - http://www.usb.org/developers/docs/devclass_docs/DFU_1.1.pdf, along with this patch not from u-boot's implementation - http://lists.denx.de/pipermail/u-boot/2011-November/108003.html.

Looking at the source code, dfu-util will call libusb_get_string_descriptor_ascii() to receive a string descriptor in unicode. I am not sure what index must be set but the line of code that finds it is here - https://github.com/Stefan-Schmidt/dfu-util/blob/fdc72a8a738994dc72cfd63ff08f690e5145277e/src/main.c#L193.

From the looks of the line, and examining the libusb_get_config_descriptor_by_value() code it returns a [http://libusb.sourceforge.net/api-1.0/structlibusb__config__descriptor.html](libusb_config_descriptor) struct, which then has an array of [http://libusb.sourceforge.net/api-1.0/structlibusb__interface.html](libusb_interface) structs, each of which has an array of [http://libusb.sourceforge.net/api-1.0/structlibusb__interface__descriptor.html](libusb_interface_descriptor)'s (altsettings), which then have an index. This index is what is used to query the string. As far as I can tell there is no string field in the struct, so I am not sure where it is stored, perhaps the extra field is of use... Looking at this log it seems like the iInterface somehow has a descriptive string to go with it - https://github.com/cjheath/dfu-util/blob/99ddc0c15021065512dca5c98ce57bec79513261/device-logs/stm32f4discovery.lsusb.

It would be helpful if I knew more about how string descriptors worked in the usb standard... Just did some searching. According to this page, http://www.usbmadesimple.co.uk/ums_4.htm, iInterface is the "Index of string descriptor describing interface - set to 0 if no string". That makes a lot more sense. In that case I believe this line must be updated - https://github.com/kiibohd/controller/blob/master/Bootloader/dfu.desc.c#L53, and a string descriptor must be added somewhere. Looking for other string descriptors my best guess is in the string_descs array found here - https://github.com/kiibohd/controller/blob/master/Bootloader/usb.h#L140. USB_INIT_DEVICE does not seem to be used anywhere but it seems like the best place. I'll give it a shot and report back.

Happy debugging.

smasher816, 2015-03-08:
Got my buspirate wired up and tested this thing out. Turns out it didn't read my ID=4 I added, but using id 3 (serialno) worked fine. This lead me to find out that the defines in usb.h do not seem to be used, instead the string must be added to dfu_device_str_desc in dfu.desc.c. After doing that and adding a define to _buildvars.h I could successfully set iInterface to the index of the new string and have it show up as the name in dfu-util --list. See here: http://ix.io/gNP

If you say what you want the name to be I will send a pull request over. I could send one now but then you would probably end up changing it anyways. Also, it might be worthwhile to look at usb.h and see if some of that stuff is actually needed (it appears it is not).

haata, 2015-03-08:
Cool. Awesome work.

Probably the easiest thing to name it would be "Kiibohd DFU". Usually for names like this I put it in one of the CMake build files (through _buildvars.h). Probably Lib/CMake/arm.cmake would be the ideal place for it.

smasher816, 2015-03-09:
Created a pull request using your preferred name and define location. Tested working on my infinity, but feel free to review the code. It's a pretty simple change overall, but the research going into it is what made it more difficult :)

PR #25 (2015-03-09): Set DFU bootloader name
Fixes #22

Each (dfu) interface may have more than one a ltsetting each with their own index and name. According to the DFU_1.1 pdf section 4.2.3, "* Alternate settings can be used by an application to access additional memory segments. In this case,
it is suggested that each alternate setting employ a string descriptor to indicate the target memory
segment; e.g., 'EEPROM'." Whether or not we end up using multiple memory segments it is still good to have a descriptive name incase there are other dfu devices connected.
commit dbf2b1ca (2015-03-09): Set DFU bootloader name

Fixes #22

Each (dfu) interface may have more than one a ltsetting each with their own index and name. According to the DFU_1.1 pdf section 4.2.3, "* Alternate settings can be used by an application to access additional memory segments. In this case,
it is suggested that each alternate setting employ a string descriptor to indicate the target memory
segment; e.g., 'EEPROM'." Whether or not we end up using multiple memory segments it is still good to have a descriptive name incase there are other dfu devices connected.
commit ef8a8c5d (2015-03-09): Set DFU bootloader name

Fixes #22

Each (dfu) interface may have more than one a ltsetting each with their own index and name. According to the DFU_1.1 pdf section 4.2.3, "* Alternate settings can be used by an application to access additional memory segments. In this case,
it is suggested that each alternate setting employ a string descriptor to indicate the target memory
segment; e.g., 'EEPROM'." Whether or not we end up using multiple memory segments it is still good to have a descriptive name incase there are other dfu devices connected.

Edit: Fixed previous indentation
closed 2015-03-09 by haata