kiibohd/controller #24: Keyboard does not wake up computer

haata, 2015-03-08:
Firmware currently does not wake up a sleeping/suspended computer.

There are a few things that need to happen to make this work:
http://cache.freescale.com/files/32bit/doc/ref_manual/K20P48M50SF0RM.pdf 40.4.9

When the computer is sleeping, to wake it up this needs to be done.
USB0_CTL |= USB_CTL_RESUME;

Whenever there is a keypress (signal over usb) + usb has suspended, a resume signal should be sent.
(Likely USB0_ISTAT & USB_ISTAT_SLEEP).

rossipedia, 2015-03-08:
Does this affect all operating systems?

haata, 2015-03-08:
I think it does, this is a pretty standard usb thing.

Not that hard to add really (just have to find a good spot for the code and make sure that the suspend state is detected correctly).

kaansoral, 2015-05-09:
I also noticed this, I think this can be a shortcoming turned feature thing, I haven't figured out how to send system/consumer codes yet, but would the "Wake" code wake the computer?

If so, it's probably best to wake the computer with a specific combo

haata, 2015-08-20:
Total accident, but yesterday's update seems to have fixed the USB wakeup issues as well. lol

Tested on Mac OSX and Windows 10.

closed 2015-08-20 by haata