kiibohd/controller #28: Documentation Improvements

braidn, 2015-03-14:
Is it just my Ruby/Non C addled brain but could there be some improvements with the documentation? Especially around the build process, and what is absolutely needed to generate a .bin file? Is there room for me to pitch in and help with this?

haata, 2015-03-14:
Sure! Help is always welcome :D
Smasher has been helping move the readme to the wiki, but it would be nice to have more detailed/straightforward instructions.

It would be nice to have some infinity specific instructions. This will become much more important as I release new keyboards in the near future.
I'm still trying to figure out the right balance between power/complexity and ease of use for first-timers.

Perhaps some "quick-build" scripts are in order along with the documentation.

smasher816, 2015-03-14:
I am done moving over all the content from the readme. It have all the content that was originally there. The wiki pages should be open so feel free to clarify anything if you can - or even just add a "TODO: Could be explained better" or something like that for others to work on.

@haata I think quick build scripts would be a good idea. Would it be possible (or a good idea) to make a "Keyboards" type directory that contains the most common cmake defines. Ex: Infinity has mk20dx128vlf5 chip, MD1 scan, PartialMap macro, and pjrc usb. Then someone could run a script like `./build.sh infinity` or whatever the name of a supported board is. It would then do a build with those definitions without any extra work, which would be nice for the first timers. For the more adventurous they could always dabble in the CMakeFiles themselves as they would have to now.

haata, 2015-03-16:
I've made a first pass at the convenience compilation scripts here: 337eaa16f27856df89b06ec47e89e39f7f1976c1

Still needs a bit more work to handle the cygwin case, but should be fine for Linux and Mac.

haata, 2015-03-16:
Updated the README to be easier to read for a beginner.
a6edcce5f991dd1511acd55fa208578f7b063431

unphased, 2015-05-31:
I would suggest tips on how to transition kll's that were made with the web configurator into a build folder. I can build stuff  but I dont know which files to overwrite where with which other files to be able to use this. Seems like cmake is needed to generate a bin from kll's?

braidn, 2015-05-31:
It is actually quite confusing to figure out where you need to be/should be when building a proper layout file. I also have not had a ton of time to help out on Doc betterment. However I wrote up a quick how-to here: http://cloudbacon.com/2015-04-05-infinity-keyboard-info 

It will have you building layout files in no time.

unphased, 2015-05-31:
@braidn The breadcrumbs are much appreciated. Thanks.

braidn, 2015-06-01:
@unphased https://github.com/braidn/infinity. This is the build file I am using and the corresponding layouts.