kiibohd/controller #30: /usr/lib/gcc/arm-none-eabi/4.9.2/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory

PaBLoX-CL, 2015-03-26:
```
./infinity.bash
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx128vlf5
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- Detected Scan Module Source Files:
Scan/MatrixARM/matrix_scan.c;Scan/MD1/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c
-- Detected Output Module Source Files:
Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/bin/git (found version "2.3.4") 
-- Found Ctags: /usr/bin/ctags (found version "5.8") 
-- Detected Layout Files:
/home/pablo/code/kiibo/controller/Scan/MatrixARM/capabilities.kll
/home/pablo/code/kiibo/controller/Macro/PartialMap/capabilities.kll
/home/pablo/code/kiibo/controller/Output/pjrcUSB/capabilities.kll
/home/pablo/code/kiibo/controller/Scan/MD1/defaultMap.kll
/home/pablo/code/kiibo/controller/kll/layouts/md1Overlay.kll
/home/pablo/code/kiibo/controller/kll/layouts/stdFuncMap.kll
/home/pablo/code/kiibo/controller/kll/layouts/hhkbpro2.kll
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pablo/code/kiibo/controller/Keyboards/template
Scanning dependencies of target kiibohd.elf
[  5%] Building C object CMakeFiles/kiibohd.elf.dir/main.c.o
In file included from ../../Lib/mk20dx.h:57:0,
                 from ../../Lib/Interrupts.h:34,
                 from ../../Lib/MainLib.h:32,
                 from ../../main.c:25:
/usr/lib/gcc/arm-none-eabi/4.9.2/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
CMakeFiles/kiibohd.elf.dir/build.make:68: recipe for target 'CMakeFiles/kiibohd.elf.dir/main.c.o' failed
make[2]: *** [CMakeFiles/kiibohd.elf.dir/main.c.o] Error 1
CMakeFiles/Makefile2:95: recipe for target 'CMakeFiles/kiibohd.elf.dir/all' failed
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2
Makefile:75: recipe for target 'all' failed
make: *** [all] Error 2
Firmware has been compiled into: 'template'
```
Is this a bug within the `arm-none-eabi-gcc` package?

PaBLoX-CL, 2015-03-26:
Sorry, premature "Enter" XD.

I was using arch linux, I solved installing the following package:

    pacman -S --needed arm-none-eabi-newlib

Should be this added to the wiki?

closed 2015-03-26 by PaBLoX-CL

haata, 2015-03-26:
Yep, probably should be in the wiki. :D

smasher816, 2015-03-26:
Added to the arch linux page. I'll look up packages for Ubunutu/Mac and add them in later.

bijanbina, 2015-07-12:
Thanks That realy work

tigercoding56, 2023-04-24:
> Sorry, premature "Enter" XD.
> 
> I was using arch linux, I solved installing the following package:
> 
> ```
> pacman -S --needed arm-none-eabi-newlib
> ```
> 
> Should be this added to the wiki?

thank you ,
i ran across this issue building another opensource project (infinitime)