kiibohd/controller #31: Sometimes the keyboard follow X configuratio

PaBLoX-CL, 2015-03-27:
I have my laptop setup to a latam layout, set through xorg.conf. Sometimes, the infinity keyboard tries to act as one too, leading to weird situations :/. Right now for example, the Infinity has the "ñ", making a lot of keys just dissapear (in latam layout there are 3 keys to the right of the `l`:

<kbd>ñ</kbd> <kbd>{</kbd> <kbd>}</kbd> 

When the infinity acts a latam layout, I lose the <kbd>}</kbd>.

Any idea what could be happening? (reflashing, removing/reinsertng the keyboard doesn't work).

haata, 2015-03-27:
Hmm. There could be a bunch of reasons for this.

* Special USB descriptor case (Linux hack to fix some broken keyboards)
* Not using the intended { } keys
* Odd Xorg bug

Unless there is a problem with the USB descriptors I don't think it's a keyboard problem. Try using xev to see if what you press is what you expected to happen. The infinity keyboard also has a debug cli that you can use (outputDebug command).

I wonder if hasu/tmk firmware has similar issues?

smasher816, 2015-03-27:
Have you tried keeping the infinity in a standard qwerty layout, and changing the host system to latam? Or the opposite - latam infinity, default (qwerty) system layout.

I believe if you set both to latam then keys will be remapped incorrectly. This is because windows, Linux, and other OS's expect the keyboard to be qwerty and remap the keys on that expectation (ex: q->a for azerty).

PaBLoX-CL, 2015-03-27:
This is strange, because somtimes both keyboards (the one for my laptop) and the infinity doesn't follow the configuration from X. Of course I can swap the layouts from the system, but kind of misses the point of a programmable keyboard. For instance, during the morning, my laptop was working as a laatam layout while the inifnity was using the us layout.

Anyway, the "Fn" buttons, keep working fine (I mean the other layout), the "Function" button doesn't produce a keycode, but the other buttons are working.

This is a bit annoying, because sometimes I do want the caps_lock, and the position this layout uses is nice (bottom right), but since I have set `ctrl:nocaps` I lost the button :(

haata, 2015-03-27:
Yeah, there are a lot of limitations to the USB spec. I have a long term goal of being able to output unicode but as it stands now it's up to OS implementations to do special characters.

So, there might be something else that can be done. I know Mac OSX can detect whether a keyboard is a specific language using the Language field in the USB descriptor. I'm not sure if Linux keys on this, but I think evdev might be able to.

https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/arm/usb_desc.c#L405 is the specific line.
If changing this value helps then I can add it as a KLL define to be more convenient.

PaBLoX-CL, 2015-03-28:
I changed it to `33` (US) according to the documentation about the USB spec. `lsusb` correctly finds the `bCountryCode`.
```
Bus 003 Device 008: ID 1c11:b04d  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1c11 
  idProduct          0xb04d 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          125
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      63
         Report Descriptors: 
           ** UNAVAILABLE **
…
```

Obviously the problem is from Xorg. I don't know the internals, but afaik, the keyboard is configured through `evdev`... I don't know if it makes sense, but maybe we could make the xserver to ignore the keyboard?

haata, 2015-08-19:
I've added a kll define to change the country code (it can be overridden in your custom kll keymap file).

https://github.com/kiibohd/controller/blob/20b62afb9a05a64b1f15c6329866600913775eea/Output/pjrcUSB/capabilities.kll#L22