kiibohd/controller #34: 'make' fails with cli.c undefined reference errors (screenshot attached)

fucema, 2015-04-08:
Ubuntu 14.10 mini, in a Virtualbox VM x86_64
Linux kernal 3.16.0-33-generic

I've followed build documentation here but am failing with several errors related to cli.c. Any help provided is appreciated!

[image]

haata, 2015-04-08:
Interesting. There were some recent code changes in the cli, I'll give it a check.

haata, 2015-04-08:
Just pushed a change. Sorry about that, haven't compiled from master in a few days.

fucema, 2015-04-08:
Yeaaaaaaaa baby, thanks for the fix! I was going mad trying to figure out what I was doing wrong.

closed 2015-04-08 by fucema