kiibohd/controller #35: SYS/CON codes from KLL doesn't work

kaansoral, 2015-05-09:
I hope I'm missing something obvious but the SYS/CON definitions doesn't work

<trigger> : CON”Play”;

For example

The Consumer/System code table is impressive, can't wait to try them, yet couldn't figure out how to do it yet

kaansoral, 2015-05-09:
sysCtrlOut / consCtrlOut / usbKeyOut functions should be used instead, for those who stumble onto this issue

so far I'm loving the firmware, it requires a bit digging, but it's been rewarding so far

haata, 2015-05-09:
Apologies on the CON"Play" part. I just finished that part of the spec last weekend. Haven't finished the compiler part (almost there though :D). I'll update this once it's working.

Thanks a lot for the bug reports.

kaansoral, 2015-05-09:
No problem, it was a joy to use the firmware, once the bugs are solved, it will probably flourish

One small bug that I noticed is: changing the defaultMap name on the .bash file doesn't work, the Scan/MD1 one is used instead (I added my custom .kll's to the kll/layouts folder, only the default one isn't recognized)

haata, 2015-05-10:
I've just pushed the update to the kll compiler. Make sure you update the kll directory git repo.
Let me know if you have any issues.

closed 2015-05-10 by haata