kiibohd/controller #36: Result sequence/combinations doesn't work

kaansoral, 2015-05-09:
I've tried U"F13"+U"I"+U"F13" in various combinations, none worked, sometimes one F13 registers only once

The U["Q"-"Y"],U["Enter"] example only prints Y and enters

Overall, I've tried various things, however I couldn't implement basic macros, with neither of sequnces/combinations

(F13 is my language change key, I'm trying to add special characters to my layer, by shifting the language with a macro and restoring it later on)

After reading the documents, it's unclear how sequences/combinations are supposed to work / their differences etc.

haata, 2015-05-09:
So, a range like U["Q"-"Y"] is usually for a trigger part of the macro rather than the result.

So I'd think U["Q"-"Y'], U["Enter"] : U["Q"]; to be any key from Q till Y then Enter to result in Q.

For the output, a range makes less sense, it would be up to the compiler which key in the range to output from that trigger.

Perhaps give me an explicit example and then I can let you know how to build it (or whether it's actually possible using the USB spec).

kaansoral, 2015-05-09:
What I want is: (this is on a function layer)

U"I" -> F13 keydown + F13 keyup + I keydown + I keyup + F13 keydown + F13 keyup

F13 switches the language, I becomes the local version, second F13 restores the language to english

I tried
U"I" : U"F13",U"I",U"F13"
U"I" : U"F13"+U"I"+U"F13"

none worked, one of these half works

haata, 2015-05-09:
So
U"I" : U"F13",U"I",U"F13";
is what you want (that's a sequence).

Now, depending on the program, this may actually be sending the key commands too quickly. This works great for typing sentences, but sometimes programs can't handle it. I am working on a time delay mechanism as well that may help out.

(I'm not discounting bugs, because that may be the case as well).

kaansoral, 2015-05-09:
The osx system handles the F13, so one would expect it to act fast

U"I" : U"A",U"B",U"C",U"D";
U"U" : U"F13",U"I",U"F13";
U"Y" : U"F13";
U"T" : U"X",U"O",U"X";

Everything except the U"U" works, so it's most probably speed like you suspected

unphased, 2015-06-14:
I agree it would be great if we can specify the milliseconds to delay and insert them into sequences. Seems like a pretty necessary part of any macro system. 

I bet you can find out what is causing the problem by using a key scripting program to fire off those F13 keypresses and see if it stops functioning at a particular speed.