kiibohd/controller #37: Mac Issues

kaansoral, 2015-05-09:
I'm starting this thread to discuss mac issues

I have this in my function layer: U"Esc":consCtrlOut(0x032);
Holding it a bit sleeps the computer

I noticed a mac issue after using this sleep method, it's probably unrelated, but I just want to check whether other's experience it too, normally osx should ask for a password only after 4 hours in my system, however after testing this method for a bit, it started to always ask for a password, a restart fixed the issue

After the restart, I noticed that the keyboard doesn't work at the password screen, don't know what's the cause of this, re-connecting it didn't seem to make it work

haata, 2015-05-09:
Hmm, the keyboard may have gone into a locked/unknown state. I'll have to see if I can reproduce this.

flipjs, 2015-05-27:
I also have the same issue. When booting up OS X (MacBook Pro (Retina, 13-inch, Late 2012) - Yosemite), it seems the infinity keyboard is not detected. I can't type on password screen. I tried waiting for about 10 seconds or so, in case there is a delay detecting the keyboard, still no go. After logging in using macbook's keyboard, that's the time I can use the infinity keyboard.

For what its worth, here's the message/info after flashing the firmware:

<pre>
Match vendor ID from file: 1c11
Match product ID from file: b007
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        25436 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
</pre>

haata, 2015-05-27:
I believe this may be related to the Sleep/Wakeup issue, but it might be Mac OSX doing some sort of lock-out for untrusted devices. I'll try to look into it further.

kaansoral, 2015-05-27:
My device wakes up well from sleep, there is only the 10 second delay, yet I also experience this startup issue specifically

I also had a hunch that something funky was going on related to the untrusted devices, when I first connected the infinity keyboard, the osx started locking up after each sleep, while it should lock up only after 4 hours

unphased, 2015-05-31:
@kaansoral For me I noticed (in the absence of any USB or other devices attached) that often it will prompt the login after a duration that is known to be shorter than the specified lockout duration. 

What I discovered was if you just hit Esc and then wake the machine again with some other key, boom you are past the login. It seems to be some kind of OS X bug.

flipjs, 2015-05-31:
For what its worth, I have no issues waking up may macbook pro (late 2012 yosemite) from infinity keyboard. My only is issue is from when booting up the machine, I cant use the keyboard yet. It could be OS X, because my deathadder mouse is not working properly either. I can move the mouse pointer, but it goes in opposite direction.

kaansoral, 2015-05-31:
Interesting, are you sure it's a deep sleep and not a monitor sleep etc?
Mine also wakes the yosemite up from light sleeps, but once everything shuts down, I use the mouse for waking

kaansoral, 2015-06-14:
I started experiencing lost first characters, I don't know whether this is specific to mac or an issue in general

Normally, when the os/keyboard wakes up and when you type "something"

The response is usually:
2-5 seconds delay + "something"

Lately, it's sometimes:
2-5 seconds delay + "omething"

I would rather have:
no delay + "something"

haata, 2015-06-14:
Lol yep, I want no delay too.
I'll have to see whether it's to do with the keyboard or just an annoying
Mac OSX wake up thingy.

unphased, 2015-06-14:
OS X waking up is always flaky for me, with the built-in keyboard. I don't expect it to improve.

unphased, 2015-06-14:
So I find it quite interesting that I can type into the Infinity keyboard immediately after it's flashed and 10 seconds later the chars land

haata, 2016-05-17:
Fixed a bunch of Mac OSX issues.

https://github.com/kiibohd/controller/commit/8fbc8d535f6fd858fb549c4f85c83ee1791f77d7
https://github.com/kiibohd/controller/commit/ba58010a7582b11b5fff2b995eb406e27fa44ecf