kiibohd/controller #38: Feature Request: Triggers for *exact* response

kaansoral, 2015-05-27:
Current behaviour:
"Alt"+"A":"B" becomes "Alt"+"A"+"B"
"Alt"+"Shift"+"A":"CMD"+"B" becomes "Alt"+"Shift"+"A"+"CMD"+"B"

What it should be:
"Alt"+"A":"B" becomes "B" (Alt cancelled and re-instated later on)
"Alt"+"Shift"+"A":"CMD"+"B" becomes "CMD"+"B" (Alt+Shift cancelled + re-instated later on)

What I'm ultimately saying is, triggers just trigger additional actions, yet there is a need for altering behaviour to create pure modification combos

I want "Shift"+"Esc":"~" to become a Tilde, yet the "Esc" is never cancelled, messes up in the Terminal
I also want to merge my function layer with the Command modifier, so Command+C should be Copy, yet Command+[ should be F14 (Command modifier cancelled with a custom *exact* trigger and F14 sent instead)

None of these are possible with the current feature set

AUmrysh, 2015-05-27:
I think it's related to this issue:  https://github.com/kiibohd/controller/issues/18

kaansoral, 2015-05-27:
Yes, it definitely is, he also experienced the same issue as the feature doesn't exist

haata, 2015-05-27:
I'm working on the spec to include these features, already on the second
draft.

kaansoral, 2015-05-27:
It's great to hear, can't wait, thanks :)

kaansoral, 2015-06-11:
By the way, is there an ETA for these new features?
I'm a bit anxious to finalise my reflexes, otherwise they have been very jumbled lately
Thanks

haata, 2015-06-11:
Currently there isn't an ETA.
I've basically finished the spec though, will be released soon. At which point I can begin implementing.

I need to finish the changes required to ship the Infinity ErgoDox and another upcoming keyboard first. But after that I can start looking at bug reports and feature requests again.

kaansoral, 2015-07-18:
Hi

I'm considering adding this feature myself, I've briefly inspected the code, it's very well commented, although I didn't take the time to understand everything - only the skeleton - so at this point I don't know how things work

If the input logic is completely separated from the output logic, I think adding these kind of features would be extremely easy - also adding delays to the routines would be possible too

The input routine would schedule/queue exact sequences to be sent
The output routine would parse these sequences, wait if needed, press and release modifiers based on the differences of sequences

Lets say the sequence is A B Shift+A Shift+B X Shift+A - it would momentarily pull and re-engage the shift before/after X

In this scenario, the U"Shift"+U"X" -> U"X" exact response would be possible

Once again, I didn't assess how things work with the infinity keyboard, but I briefly considered how I would implement a keypress logic to support what I want, and this seemed simple/possible