kiibohd/controller #39: Layers locked from another layer don't take priority.

mattventura, 2015-05-28:
Using this base layer + 2 partial layers to demonstrate the issue:
[image]

If I press 'asd', it types 'asd' as expected. 
If I press Ctrl then type it again, it types 'qwd' as expected. 
However, if I press 'z', then type 'asd' again, it types 'qwe' when 'fwe' would be expected. 

Debug CLI, first with neither layer on, then with layer 1, with both, and back to layer 1 only:
[image]

mattventura, 2015-05-28:
Also this doesn't happen with only layerLock, it happens with layerShift and probably layerLatch as well.

haata, 2015-06-20:
Sorry about the delay.
I've added a new debug option to help with this 99098fb2d6614f0c3538f67d61ff13251a04c741
It's called layerDebug

If you still have the issue would you mind attempting to change layer states with it on?
It'll show you all the changes.

mattventura, 2015-06-22:
Everything seems to work fine in terms of the layer stack. If I lock Fn1, then lock Fn2, layerDebug correctly shows 0, then 0:1, then 0:1:2. But if both layer 1 and layer 2 have a mapping for the same key, layer 1's takes priority. If the layer stack is 0:2:1, then layer 2 takes priority if there's a conflict. 

It looks like the partialMap code (specifically macro.c:365) starts at the bottom of the stack and stops when it finds a match. Shouldn't it be the other way around?

haata, 2015-06-22:
Basically in the case of 0:2:1, first layer 1 is checked, if nothing is defined, then check layer 2, and if nothing there is defined check layer 0.

I guess I could have wrote it as 1:2:0, but it's easier to visually check if it grows to the right.

mattventura, 2015-06-22:
Is that what's supposed to happen? It seems to work the opposite way, where if the layer stack is 0:1:2, then the priority is 1 > 2 > 0. In the case of 0:2:1, it's 2 > 1 > 0.

haata, 2015-06-22:
Oops, probably traversing the stack backwards. 0 isn't part of the actual stack because it's always the fallback. I'll try to fix this tonight.

haata, 2015-06-23:
Hopefully fixed with 216b1513026e45e44794f15cfa4d02ff1c9d49f4

mattventura, 2015-06-23:
I'm confused. Previously, if I activated layer 1, then layer 2, it would display 0:1:2 as the layer stack, but it would function as 1 > 2 > 0. Now, it displays 0:2:1, and functions the same. 

Shouldn't activating layer 1 then layer 2 cause layer 2 to take priority? I think now the debug output matches the actual behavior, but the behavior itself is bugged.

haata, 2015-08-22:
Sorry this has taken me so long.

Was browsing the Macro code when doing the Interconnect (Infinity Ergodox) and noticed the layer evaluation code was processing the layers backwards (everything else was correct).

Super easy fix. Should be working now.

51486bc4e16830c16c30f696b3a4f84c93a5baf8

closed 2015-08-22 by haata