kiibohd/controller #40: Media keys don't work after a reboot. Possibly stuck in bios mode?

dumle29, 2015-05-29:
So I've been having issues with mediakeys deciding not to work. I initially thought it was from booting in to windows, which I didn't quite understand, so I tried some more things, and it turned out, that a restart would have the same effect.

I would flash the keyboard with a .bin, and media keys would work fine in Linux, but after a reboot they wouldn't work. Un-plugging, and replugging the keyboard however, will make them work again.

This made me thing that the keyboard might be stuck in bios compatible mode, and a replug takes it out of that.

DavidK-M, 2015-12-07:
Hi There,

I might have a similar problem... or not!

After re-boot my left-hand board's shift/alt/ctrl keys don't work - the left-hand board is a slave of the right-hand board that is plugged into the computer. To get the keys to work again I un-plug/plug the board in.

Best regards,

David.

haata, 2015-12-07:
Interesting. Would you mind providing me with your computer specs? OS and version specifically.

Also, which board do you have to re-plugin? The slave or the master?

DavidK-M, 2015-12-07:
Hi Haata,

Thanks for taking interest.

The computer is running Windows 7 Ultimate (Service Pack 1) (64 bits), it only has USB2.0 ports.

It is the master that I unplug (the connectors for the connection between the two boards being "fragile").

I can connect to the serial port and talk to the firmware directly, "version" gives me this information:

: version
 Revision:      7b7a55899f392ebb7f615fd1801aaaa3dcc3f738
 Branch:        master
 Tree Status:   Clean

 Repo Origin:   https://github.com/kiibohd/controller.git
 Commit Date:   2015-10-18 17:54:41 -0700
 Commit Author: Jacob Alexander <haata@kiibohd.com>
 Build Date:    2015-11-02 20:45:33 +0000
 Build OS:      Linux-4.1.5-x86_64-linode61
 Architecture:  arm
 Chip:          mk20dx256vlh7
 CPU:           cortex-m4
 Device:        Keyboard
 Modules:       Scan(MDErgo1) Macro(PartialMap) Output(pjrcUSB) Debug(full)
 Unique Id:     8CE1000067DA00284300E28464E45
:

(So version has given me your name, thanks Jacob! It's really cool being able to talk to the keyboard over a serial console - it reminds me of the PDP11 (LSI11/23) where you could talk to the debugger built into the CPU itself over a serial port :-D )

I think that this is appropriate information, if you need any more don't hesitate to ask. 

Best regards,

David Kerr-Munslow.

haata, 2015-12-08:
Can you check the status of kbdProtocol when this happens?

My suspicion is that it'll be set to 0 instead of 1

DavidK-M, 2015-12-14:
Hi Jacob,

You are right the kbdProtocol is set to 0.

Just unplugging the left hand side from the master does not change things. (Still no shift/ctrl/alt keys and the kbdProtocol remains set to 0).

So it appears that the keyboard remains in the "boot protocol"? The procedure is that I "switch off" my machine for the weekend the keyboard remains powered(Windows shutdown, but there is still some power to the motherboard which supplies power to USB devices so that a keyboard can wake the system up, I presume) . Everything was working. When I power up the machine again on the Monday the shift keys on the slave keyboard aren't working.

Many thanks,

David Kerr-Munslow.

aphistic, 2016-03-07:
Just wanted to drop by here and mention that I'm having the same (or a very similar) problem when I reboot my Arch linux machine: Linux aphistic 4.4.3-1-ARCH #1 SMP PREEMPT Fri Feb 26 15:09:29 CET 2016 x86_64 GNU/Linux.  It's a Dell Precision T1700.

I have my right side as a slave to the left side and when I reboot, the shift key on the right side stops working.  I can see the key events when i use xev, so I know it's getting through but it just doesn't do anything.  If I disconnect the two halves it doesn't fix it and when i disconnect the computer from the right half (by unplugging the keyboard side, not the computer side) it doesn't fix it either.  However, if I plug the right and left side into the computer on their own the shift starts working again.  If I plug it back into the right side it stops. At that point I usually end up reflashing the right side and it starts working again. Next time it happens I'm going to try unplugging from the computer end of the keyboard to see if that works.

haata, 2016-03-08:
Thanks Erik. So my current thinking is that for some reason the
interconnect gets into a bad state.

I haven't tested it yet, but in theory, you can issue a 'connectRst'
command in the debug cli (
https://github.com/kiibohd/controller/wiki/Debugging) and it should clear
things up.
I'd also check to see if the keyboard is in 6KRO mode as well
'kbdProtocol'. 1 is NKRO and 0 is Boot mode.

I blame my hating to reboot as the cause of this bug :P

aphistic, 2016-03-08:
Haha, I know what you mean. I did my once a month or so update and reboot and all of a sudden my Shift key stopped working.  I thought it was something in the updates for a bit! I'll give the connectRst a try tomorrow at work and let you know what happens.  If there's anything else I can do to help, let me know!

aphistic, 2016-03-08:
kbdProtocol was indeed set to 0 after a reboot.  I tried doing a connectRst but that didn't seem to work.  Once I did a restart (in the debug console) it did, though.

haata, 2016-03-08:
Yeah, the restart is a full software restart of the keyboard firmware
(there's a chance some variables don't get re-initialized so it's not quite
as good as an unplug/replug).
Hmm, so the interconnect is probably functional (which is good, it's harder
to fix :P).
It's interesting that the keyboard is still in boot mode. That means your
BIOS/OS is not doing what it's supposed to be doing. The keyboard has to be
told to go into boot mode (it defaults to NKRO). This means that either the
OS or the BIOS is asking for boot mode. If it's the BIOS, then that's ok,
but it seems like it's handing off the descriptor information to the OS. I
think there's a BIOS setting for this on some computers.

I have another bug open to show on the LCD if the keyboard is in 6KRO mode,
which should help find these issues more easily. Still, the keyboard should
work in 6KRO mode (so it's a firmware bug regardless).

aphistic, 2016-03-08:
Ah, I just thought of something. It's probably good to note that I'm using UEFI instead of legacy BIOS.  Maybe there's a problem with rebooting UEFI systems?

haata, 2016-03-08:
Hmm, that might have something to do with it. But yeah, not sure at this
point. Good to know.

DavidK-M, 2016-05-02:
Hi there,

I have just tried the  'connectRst' and it does not seem to help.

The left hand (slave) keyboard shift/ctrl/... modifiers are still not recognised :-( 

I still have to unplug/plug-in the keyboard.

Cheers,

David.

haata, 2016-05-02:
Have you tried the 'restart' command?
This should completely re-initialize the USB bus. If the problem is on the
slave side, I may have to look a bit further.

DavidK-M, 2016-06-28:
The "restart" command does the trick...

I wonder if it is possible to map the "restart" command to a key-mapping ...?

haata, 2016-06-28:
Try mapping a key to 6/N-KRO on the configurator. This should have the same effect until I'm able to fully fix this.

haata, 2016-07-29:
Not a 100% fix, but this should resolve the non-working media keys.
https://github.com/kiibohd/controller/commit/9309889fad2fd31518eff6fded681d1281d66dd9

patrickdepinguin, 2018-08-20:
I'm now experiencing this issue too. I'm using hibernate on Linux. When the PC comes back into the OS after hibernation, some special keys do not work. I checked via the debug console and can see the keyboard is in 6KRO mode. If I unplug/replug I get correct NKRO mode.

Let me know what I can do further to help with this. It's bothering me enough to want to spend some time for it.

haata, 2018-08-20:
Hmm, how recent is the firmware?

haata, 2018-08-22:
I did do some work a few months ago simplifying USB descriptors so it might be worth trying.

https://github.com/kiibohd/controller/commit/bac8dec83fc56cdf20dcc6f15629045145e4c413

patrickdepinguin, 2018-08-22:
I updated repos to 49a21d799575c3790da766ebe18b8dd32bfe4fda and kll@255ece0ee216796af4b9a898644c8edd94e146c2 .
After flashing the keyboard, I enabled outputDebug via the debug interface, verified it was NKRO, then triggered a hibernation cycle. When the system was back up, I reconnected to the debug interface, and I saw 6KRO was active.

So, problem is still present on latest. Let me know what I can do...