kiibohd/controller #41: How do I bind multiple modifiers in a single key?

flipjs, 2015-05-29:
Say for example, I want RALT to become LCTRL+LSHIFT+LALT. Can I do this:

U"RALT" : U"LCTRL+LSHIFT+LALT";

Is there a documentation somewhere on how to bind keys? Thanks.

haata, 2015-05-29:
Yes, this should already be supported. You can take a look at the spec. https://www.overleaf.com/read/zzqbdwqjfwwf

This should work though.
U"RALT" : U"LCTRL" + U"LSHIFT" + U"LALT";

flipjs, 2015-05-30:
It didn't work. I edited MD1-Standard-0.kll both in root and kll folder:

Name = "MD1";
Layout = "Standard";
Version = "0.2";
Author = "HaaTa (Jacob Alexander) 2014";
KLL = "0.3";
Date = "2014-09-14";
Generator = "KIICONF 0.1";

U"FUNCTION1" : U"RCTRL";
U"FUNCTION2" : layerShift( 2 );
U"RGUI" : layerShift( 2 );
U"RALT" : U"LCTRL" + U"LSHIFT" + U"LALT";
U"FUNCTION3" : U"LEFT";
U"FUNCTION4" : U"RIGHT";

The result is still Right Alt (or Option R, Im on mac osx).

flipjs, 2015-05-30:
I think I'm doing it wrong. I tried binding it the same as RGUI:
U"RGUI" : layerShift( 2 );
U"RALT" : layerShift( 2 );

Its still producing 'Right Alt'.

Do I need to recompile kiibohd.dfu.bin? All I was doing is edit the MD1-Standard-0.kll and then "dfu-util -D kiibohd.dfu.bin".

haata, 2015-05-30:
Most definitely you'll need to recompile.
KLL files get converted to C code, then the firmware (dfu.bin file) gets
generated.

flipjs, 2015-05-30:
How do I recompile it? Could you link me to documentation. Also, do I need all the source files here on github? All I have is the zip file generated from http://configurator.input.club/ and work from there. Thanks haata.

haata, 2015-05-30:
https://github.com/kiibohd/controller
https://github.com/kiibohd/controller/tree/master/Keyboards
https://github.com/kiibohd/controller/wiki

flipjs, 2015-05-30:
Thanks haata!

closed 2015-06-30 by flipjs