kiibohd/controller #42: Can't get serial to work

unphased, 2015-05-31:
Hi, I built my infinity keyboard today and I'm on a Mac trying to receive serial output. I do have a `/dev/tty.usbmodem1423` that exists whenever the keyboard is plugged in, and also the keyboard does work when plugged in, the orange LED does light if I press the flash button, but I am unable to receive any output when I run `screen /dev/tty.usbmodem1423`. 

In the wiki, `TODO: Add a summary of useful commands one might want to use in the serial console.` I was wondering maybe someone can elaborate on that? I read somewhere I should be able to make the LED come on if i type `led` into the serial, but this does not happen. So it seems like I have not actually been able to establish a serial connection here.

haata, 2015-05-31:
When the orange LED is on the keyboard is in "Bootloader Mode" so the USB serial port wont function.

Just in case there are issues with the default image that may have been fixed. I would suggest going to the [configurator](http://configurator.input.club) and just click on `download firmware` without changing anything. That will give you a basic layout.

Once you connect to the USB serial port press enter a few times and you should get a prompt.
The `help` command should give you all the basics (for more detailed command info you'll have to look into the code).
If you want to test to see if the serial port is working (but can't see what you are typing), type `led` then press Enter. The led should toggle.

unphased, 2015-05-31:
Aha. I was attempting to do this from within tmux, and I guess there are some low level issues, I am now able to get the serial working fine once attempt to use screen from a regular terminal instead of a tmux hosted pseudoterminal. `led` command indeed does work! Sweet.

Thanks for the tips. I already know that the configurator will not satisfy me, I intend to customize the snot out of this thing.

closed 2015-05-31 by unphased