kiibohd/controller #43: Interested in accessing serial from pins

unphased, 2015-05-31:
I am wondering if there are pins I can use to access the serial interface? I am thinking about having a separate MCU with its own display or colored LEDs that I can use to show the layer and lock statuses that the keyboard is in. You know, because infinity keyboard only has a single LED, and it's on the bottom and not really bright.

haata, 2015-05-31:
The TX0 and RX0 is a UART device (second drop infinity). Currently this is
used as an optional debug terminal (when USB isn't working correctly). By
default it's disabled.
If you don't mind writing C code, You should be able to control PTA1 and
PTA2.

unphased, 2015-05-31:
That almost sounds like an answer in the affirmative but could you maybe elaborate a tiny bit? Say I want to drive some LEDs with a teensy or arduino, could I hook that up to the P2 header (which has TX0 and RX0) and read out key state info? Would I be able to just leverage the interactive serial command line (the one where I can type help, but i guess I'd have to write a routine for this to switch on to provide debug feedback for when layer state switches)?

haata, 2015-06-03:
So there are a few ways. One would be to add additional commands to the cli, enable the debug UART and use that to query information (as long as not done too rapidly as it's a performance hit to access the cli).

You could also design your own serial interface (more efficient protocol).

You could use the 4 GPIO (RX0, TX0, SWDIO and SWDCKL) to indicate state as well.
Basically any attempt will require at least some modifications to the keyboard firmware (I happily accept patches :D)

unphased, 2015-06-04:
Great, sounds like a handful though, i won't be able to go down this path until I can get the keyboard running in the way that I want first (with a bunch of fancy layers of course)

thjwong, 2018-12-02:
(Sorry to reply to this really old issue)

I wonder if there has been any progress in terms of some users have already connected the LEDs (for underglow or for status indicators) and modified the code to incorporate the required LED firmware to implement that function.

I have a IC60 pcb without the 4-pins next to the reset button, and I need a pointer where the LEDs can be soldered to.