kiibohd/controller #44: USB to Bluetooth adapter doesnt seem to work

unphased, 2015-05-31:
I'm quite a ways from making progress on debugging this but basically I have one of these: 

http://www.amazon.com/Nulaxy-Wired-Keyboard-Bluetooth-Adapter/dp/B00T5YXNL0

And it's not able to complete the pairing procedure...

I guess this is not an easy problem to solve and it is easier to try out more of these devices until we find one that works. I was hoping if it worked out of the box that I could just take the casing apart and embed it into the infinity keyboard and I'd have myself a bluetooth-capable infinity bent plate board. 

So I'm wondering if anyone's tried gadgets like this and experienced success.

kaansoral, 2015-05-31:
Does it work with other keyboards? Maybe the NKRO stuff is messing with it's logic

On the bright side, I've measured infinitykb's power usage, it was at 0ma with the device I was using, so if you manage to make it work, the result will be satisfactory in terms of power usage :)

haata, 2015-05-31:
You might be more successful if you set the keyboard in 6KRO mode.
There isn't an easy way (yet) from the configurator, but you can do:

U"A":kbdProtocolBoot();

This will set the boot protocol when you press a.

unphased, 2015-05-31:
Awesome i will have to give that a try!

Currently I'm just running some configurator-generated layouts, and having some problems where my layer 1 function seems to get locked on permanently... thats totally separate and not necessarily an issue yet

unphased, 2015-05-31:
I was able to add this line and it caused the 'a' key to no longer register an a. 

However it is not clear how I can independently verify that the protocol changed. For example I plug it in, wait for it to initialize, mash A a bunch of times, no A's are sent by the keyboard, but I can still from this point press a lot of keys and they all seem to register. On non-NKRO keyboards what usually happens is it "locks up" till the keys are released. That's not happening. And i tried the bluetooth pairing a bunch of times, still no dice.

unphased, 2015-06-08:
Here is the progress I made. 

```
▸ AWHostedViewer/              |~
diff --git a/Output/pjrcUSB/arm/usb_dev.c b/Output/pjrcUSB/arm/usb_dev.c
index f09540c..0504738 100644
--- a/Output/pjrcUSB/arm/usb_dev.c
+++ b/Output/pjrcUSB/arm/usb_dev.c
@@ -455,7 +455,7 @@ static void usb_setup()
                printHex( setup.wValue & 0xFF );
                print(NL);
                #endif
-               USBKeys_Protocol = setup.wValue & 0xFF; // 0 - Boot Mode, 1 - NKRO Mode
+               USBKeys_Protocol = 0; // 0 - Boot Mode, 1 - NKRO Mode
                endpoint0_stall();
                return;

diff --git a/Output/pjrcUSB/output_com.c b/Output/pjrcUSB/output_com.c
index 7a9b84b..26d98e8 100644
--- a/Output/pjrcUSB/output_com.c
+++ b/Output/pjrcUSB/output_com.c
@@ -114,7 +114,7 @@ volatile uint8_t  USBKeys_LEDs = 0;
 // Protocol setting from the host.
 // 0 - Boot Mode
 // 1 - NKRO Mode (Default, unless set by a BIOS or boot interface)
-volatile uint8_t  USBKeys_Protocol = 1;
+volatile uint8_t  USBKeys_Protocol = 0;

 // Indicate if USB should send update
 // OS only needs update if there has been a change in state
@@ -191,7 +191,7 @@ void Output_kbdProtocolNKRO_capability( uint8_t state, uint8_t stateType, uint8_
        Output_flushBuffers();

        // Set the keyboard protocol to NKRO Mode
-       USBKeys_Protocol = 1;
+       USBKeys_Protocol = 0;
 }
```

This is fine considering how little I know of what's going on, I am finally able to see non NKRO output in debug over serial. 

However, very sad to say, this made no difference and the bluetooth keyboard pairing does not work.

haata, 2015-06-08:
Interesting. I just ordered one of the devices.

Also, I think I may have contact info for the developer of the device.

unphased, 2015-06-08:
Thats fantastic, yeah I would imagine having a way to debug the bluetooth device is probably what you need