kiibohd/controller #45: Causes OS X kernel-panic/freeze/reboot

unphased, 2015-05-31:
I have not even the slightest clue about how to test this. Maybe using a logic analyzer reading the USB signals, continuously logging output to a separate computer...

Has only happened twice so far.  

The second time I was just typing. First time I don't remember.

kaansoral, 2015-05-31:
Normally osx devices keep working, so when they fail it's generally a paranoid experience

I was also using my old macbook air to test the infinitykb, probably a 2012/3 model, and during the trials, when I was working on the swithces/keyboard/case - the device was shutdown unexpectedly once

I checked the logs, yet there was no mention of any issue

Did you check the "Console" app logs?

kaansoral, 2015-05-31:
After reading about your issue, I decided to check my "Console" logs too

It was filled with: [SendRawHCICommand] ### ERROR: EnqueueRequestForController failed (err=e00002d8)
errors

When you google it - it's generally associated with sleep/wake issues

To test whether it's related, I've removed the keyboard and watched the logs again, the errors re-appeared - they are probably errors that pop up when bluetooth is disabled and there was a connected element before disconnecting bluetooth - enabling bluetooth got rid of the error logs

TL;DR: I couldn't uncover anything from the "Console" logs, these errors logs turned out to be unrelated - using the keyboard for almost a month, I didn't experience any issue to warrant search - but it's a good idea to check the "Console" to detect whether an issue exists, I will keep an eye out too

unphased, 2015-05-31:
Yeah I'll def keep an eye out. Hoping I can switch the board to being my daily driver soon, but actually when I do this, that would mean having it on my Windows 7 desktop. 

My main side-quest is to have the infinity board up and running reliably with the Nulaxy Bluetooth (that has a 2xAAA tray built in), I just need the NKRO switched off for it. Then I will use it with iOS. And I highly doubt that key codes over bluetooth could cause a system crash.

kaansoral, 2015-05-31:
Believe me anything could happen

I suggest requesting a compiled tmk_keyboard firmware for infinity from someone, you can use that for testing too, if it works, at least you would now that it's possible

I personally love the usb connection, hate bluetooth and wireless peripherals, you seem to be a person of a similar mindset, you might consider settling with a usb cable too :)

A Nerd60 is a praised keyboard for example, but I just read a review that someone was having constant drops while using nerd60 with an osx device

Anyway, let's both keep an eye out for the osx issues :)

I'm also using the keyboard through a USB 3.0 Hub, for what it's worth, if there is an electrical issue or something, that hub might be acting as a shield

unphased, 2015-05-31:
Well the thing motivating me for bluetooth is definitely not to ever use it with a full-on computer, it's to see what I can do on iOS -- no consumer bluetooth keyboard I know of comes close to supporting something as simple as mapping the caps lock button to something else. 

Good idea about the USB hub

haata, 2016-02-22:
I've added support for the Apple IPad Air (only iOS device I have). During testing I did notice a crash/reboot. But I haven't been able to reproduce it again. I suspect it may have had something to do with the suspend/resume code (which I have also fixed up a bit in this commit for Apple and Windows hosts).

https://github.com/kiibohd/controller/commit/46765e85c57e19b89754f4f9c26c8f18b35148e5

haata, 2016-05-17:
I've done a bunch more work on USB sleep/resume and bios handling.

https://github.com/kiibohd/controller/commit/8fbc8d535f6fd858fb549c4f85c83ee1791f77d7
https://github.com/kiibohd/controller/commit/ba58010a7582b11b5fff2b995eb406e27fa44ecf

haata, 2019-02-18:
I'm closing this, as I've done even more USB work since the last comment.

closed 2019-02-18 by haata