kiibohd/controller #49: Trying to build Infinity for Teensy 2.0

tybenz, 2015-06-13:
I changed the chip type in `infinity.bash`. How about the scan settings? I get:

```
~/src/kiibohd/Keyboards master → ./infinity.bash
-- Compiler Family:
avr
-- MCU Selected:
atmega32u4
-- CPU Selected:
megaAVR
-- Bootloader Type:
Teensy
CMake Error at Lib/CMake/modules.cmake:53 (message):
  Scan/MatrixARM does not support the avr family...
Call Stack (most recent call first):
  Lib/CMake/modules.cmake:114 (ModuleCompatibility)
  Scan/MD1/setup.cmake:14 (AddModule)
  Lib/CMake/modules.cmake:91 (include)
  Lib/CMake/modules.cmake:140 (AddModule)
  CMakeLists.txt:152 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/tybenz/src/kiibohd/Keyboards/template/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
Firmware has been compiled into: 'template'
```

How can I get this to work? Here's my modified `infinity.bash`:

```bash
#!/bin/bash
# this is a build script template
# these build scripts are just a convenience for configuring your keyboard (less daunting than cmake)
# jacob alexander 2015

#################
# configuration #
#################

# feel free to change the variables in this section to configure your keyboard

buildpath="template"

## kll configuration ##

# generally shouldn't be changed, this will affect every layer
basemap="defaultmap"

# this is the default layer of the keyboard
# note: to combine kll files into a single layout, separate them by spaces
# e.g.  defaultmap="mylayout mylayoutmod"
defaultmap="infinity"

# this is where you set the additional layers
# note: indexing starts at 1
# note: each new layer is another array entry
# e.g.  partialmaps[1]="layer1 layer1mod"
#       partialmaps[2]="layer2"
#       partialmaps[3]="layer3"
partialmaps[1]="infinity1"

##########################
# advanced configuration #
##########################

# don't change the variables in this section unless you know what you're doing
# these are useful for completely custom keyboards
# note: changing any of these variables will require a force build to compile correctly

# keyboard module configuration
scanmodule="md1"
macromodule="partialmap"
outputmodule="pjrcusb"
debugmodule="full"

# microcontroller
chip="atmega32u4"

# compiler selection
compiler="gcc"

########################
# bash library include #
########################

# shouldn't need to touch this section

# check if the library can be found
if [ ! -f cmake.bash ]; then
	echo "error: cannot find 'cmake.bash'"
	exit 1
fi

# load the library
source cmake.bash

```

haata, 2015-06-13:
matrixARM won't work with a Teensy 2.0. I did start one for avr a long time ago, but I haven't updated it in years, so it would need to be updated.

And before you spend too much time at this. The teensy 2.0 doesn't really have enough RAM to do kll right now. The teensy 2.0++ is doable (I use it with the dph cap sense controller).

Now, I do have some RAM optimizations for kll in the `simple` branch (https://github.com/kiibohd/controller/tree/simple) that might make using a Teensy 2.0 possible. However, the code still has some issues.

Basically, to proceed, you'll need to be pretty good at embedded C.

haata, 2016-01-13:
Closing this unless someone wants to take this on...

closed 2016-01-13 by haata