kiibohd/controller #50: Are USB codes the "lowest level"?

unphased, 2015-06-14:
It seems to me that the specification for keys does not allow for the freedom of defining, for example, the fact that `Shift+/` produces a `?`. 

Now this may just be my ignorance, but I am realizing only now that it is in fact up to the operating system's locale mappings to determine what to do with Shift Slash. 

I guess what this means is that essentially if one wants to make a question-mark key, that we'd have to make a trigger that results in the combination `U"Shift" + U"Slash"`, as this turns out to be the only portable way to emit a question mark.

haata, 2015-06-14:
Yes that is correct.

Now, I do have some convenience syntax in kill for ?

U"A" : '?';

The compiler just turns this in shift + / though.

kaansoral, 2015-06-14:
Yes exactly, that's why without the requested features, we can't efficiently make use of a configurable firmware / fully programmable keyboard

Also as far as I remember, the convenient syntax also fails as expected for stuff like:

U"LAlt"+U"A" : '?';

And since it only works if the language has "/ ?" as the key, it's pretty specific to english language

haata, 2015-06-14:
Yes. Mostly because USB hid uses English as the base.

It's possible to add other languages but I'd need a lookup table (I'm only
familiar with a few layouts). Basically a USB code to letter conversion
table.
Then it would be as simple as indicating to the kll compiler which language
you want.

This will fall apart if the user changes the is language though (the OS
does not tell the keyboard anything when this happens).

kaansoral, 2015-06-14:
I don't think it would be worth the effort, since the required effort seems pretty high

On the other hand, having more control over the exact key up/down sequences would be simpler and more beneficial, this is my opinion

unphased, 2015-06-14:
I think the only reasonable thing would be that we program against what USB HID provides us, which are the keycodes. There is no solving the problem of language mappings, and there is not much sense putting them into the very limited keyboard memory. 

The sad part is that without much more memory it will simply not be practical to have hardware Asian language pictogram layers

closed 2015-06-14 by unphased

haata, 2015-06-14:
So, I do want to add unicode support at some point. But this will require some sort of driver and/or a daemon that runs on the system.
It would also be limited to just outputting characters (no press/release).

Unfortunately, there are quite a few higher priority items right now, so I have no idea when I'd be able to start to look at something like this.

unphased, 2015-06-14:
Does USB (HID) even provide a way to enter Unicode at all? 

If not, then unicode would simply need a software component. That's status quo. I say, focus on the awesome hardware!

haata, 2015-06-14:
So, there is one section in the USB HID spec that hints at it, but I'm not
sure the exact purpose of it (not keyboard specific). And I have no idea if
anyone actually implements it.

http://www.usb.org/developers/hidpage/Hut1_12v2.pdf

Chapter 17 indicates Unicode but then redirects to Chapter 18 which is for
alphanumeric displays.
It might be possible to send unicode characters to the OS using the
alphanumeric display HID provision then have some sort of driver/software
support on the OS to actually deal with what happens.

Unfortunately, I've never seen a USB alphanumeric device, so I'd be going
in blind.

mattventura, 2016-03-01:
For unicode support, you could allow unicode characters to be specified as a trigger, but have different methods of actually inputting them (Ctrl-Shift-U, Alt+, and whatever OS X uses). Then method could be changed at compile time or maybe even on the fly using a capability. You could technically already do that with the current functionality of kll (even the on-the-fly switching), it would just be a mess of layers.

haata, 2016-03-01:
Yeah, using the OS specific unicode composition methods are always a fallback option.

I've started working on a spec for a sideband protocol to deal with things like unicode that are annoying to deal with and are much easier with a host side pseudo-driver. Can read about it here: https://deskthority.net/workshop-f7/hid-io-sideband-input-device-protocol-spec-t12927.html