kiibohd/controller #51: Is it possible to have a trigger activate multiple capabilities?

unphased, 2015-06-14:
The KLL spec wasn't very clear on this. 

I am envisioning this: Left corner and right corner bottom row (Function 2 and Function 4 from `defaultMap`) are both to be Shift keys for Layer 1. 

Special tweak: I want to make them lock on a layer when both are pressed. That seems easy enough. 

But now I want a different (easier) way to un-lock them. 

First off, there seems to be no "unlock layer x" capability defined. It's toggle or bust, it seems. This is slightly non-ideal because in the absence of bright-ass LED's, I am liable to want to spam keys to make sure I hit them. Like Esc gets spammed a lot. If you are a Vim user, you know what I'm talking about.

Anyway, I was thinking, I'd like to be hitting either of those corner keys once to take me out of the lock.

Along another line of questioning I have found so far that `Latch()` works like Lock but doesnt toggle. It doesnt latch back off after a key. So it seems like `Latch()` is currently (I still need to test this more) functioning as a spammable Lock ON capability. So we're still definitely missing a Lock OFF capability.

haata, 2015-06-14:
Hmm, I'll take a look at the Lock issues.

I designed the shift/latch/lock the same way OSs do it.

Lock + Shift = Original
Lock + Latch = Original
Lock + Latch + Shift = Shifted

As long as you just use Lock the behaviour should be as intended. I may add
a layer state debug mode that shows the state of each layer whenever there
is a change to any one of them (to help debug this).

unphased, 2015-06-14:
Hmm I think I didn't think my example through when I wrote it. If i set Layer 1's Function 2 to `Layer 1 Lock` (while Function 2 in Layer 0 remains `Layer 1 Shift`), then I can still do my key-spamming and it would properly leave it off.

closed 2015-06-14 by unphased

unphased, 2015-06-14:
Okay so I can confirm, with a mapping like this: 

[image]

I can use the layer state to achieve my ON/OFF button functionality. With this I can mash the key on the right side of RShift to lock the layer ON, and I can mash the layer-shift key to turn it OFF.

unphased, 2015-06-14:
Ah, nice. I did this: `S0x3E : layerLock(1) + layerShift(1); ` and it works just like I thought it would! brilliant.