kiibohd/controller #52: MD2?

unphased, 2015-06-14:
I see various MD2 source. My Infinity keyboard wasn't the first massdrop group buy, I got mine with the last one that had the bent plate. 

Anyways, I tried building with `ScanModule=MD2` and I ended up with a strange compile error: 

```
[ 77%] [ 83%] [ 88%] [ 94%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_serial.c.o                                                                                                                                                                                                                                    │······························
/usr/local/bin/arm-none-eabi-gcc   -I.. -I.    -mcpu=cortex-m4 -DF_CPU=48000000 -D_mk20dx128vlf5_=1 -Os -mthumb -nostdlib -fdata-sections -ffunction-sections -fshort-wchar -fno-builtin -nostartfiles -Wall -ggdb3 -std=gnu11 -MMD -I/Users/lust/Documents/kiibohd-controller/Scan/ISSILed -I/Users/lust/Documents/kiibohd-controller/Scan/Ma│······························
trixARM -I/Users/lust/Documents/kiibohd-controller/Scan/MD2 -I/Users/lust/Documents/kiibohd-controller/Macro/PartialMap -I/Users/lust/Documents/kiibohd-controller/Output/pjrcUSB -I/Users/lust/Documents/kiibohd-controller/Debug/cli -I/Users/lust/Documents/kiibohd-controller/Debug/led -I/Users/lust/Documents/kiibohd-controller/Debug/p│······························
rint -I/Users/lust/Documents/kiibohd-controller/Debug/full -o CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_serial.c.o   -c ../Output/pjrcUSB/arm/usb_serial.c                                                                                                                                                                            │······························
Building C object CMakeFiles/kiibohd.elf.dir/Debug/cli/cli.c.o                                                                                                                                                                                                                                                                                │······························
/usr/local/bin/arm-none-eabi-gcc   -I.. -I.    -mcpu=cortex-m4 -DF_CPU=48000000 -D_mk20dx128vlf5_=1 -Os -mthumb -nostdlib -fdata-sections -ffunction-sections -fshort-wchar -fno-builtin -nostartfiles -Wall -ggdb3 -std=gnu11 -MMD -I/Users/lust/Documents/kiibohd-controller/Scan/ISSILed -I/Users/lust/Documents/kiibohd-controller/Scan/Ma│······························
trixARM -I/Users/lust/Documents/kiibohd-controller/Scan/MD2 -I/Users/lust/Documents/kiibohd-controller/Macro/PartialMap -I/Users/lust/Documents/kiibohd-controller/Output/pjrcUSB -I/Users/lust/Documents/kiibohd-controller/Debug/cli -I/Users/lust/Documents/kiibohd-controller/Debug/led -I/Users/lust/Documents/kiibohd-controller/Debug/p│······························
rint -I/Users/lust/Documents/kiibohd-controller/Debug/full -o CMakeFiles/kiibohd.elf.dir/Debug/cli/cli.c.o   -c ../Debug/cli/cli.c                                                                                                                                                                                                            │······························
Building C object CMakeFiles/kiibohd.elf.dir/Debug/led/led.c.o                                                                                                                                                                                                                                                                                │······························
Building C object CMakeFiles/kiibohd.elf.dir/Debug/print/print.c.o                                                                                                                                                                                                                                                                            │······························
/usr/local/bin/arm-none-eabi-gcc   -I.. -I.    -mcpu=cortex-m4 -DF_CPU=48000000 -D_mk20dx128vlf5_=1 -Os -mthumb -nostdlib -fdata-sections -ffunction-sections -fshort-wchar -fno-builtin -nostartfiles -Wall -ggdb3 -std=gnu11 -MMD -I/Users/lust/Documents/kiibohd-controller/Scan/ISSILed -I/Users/lust/Documents/kiibohd-controller/Scan/Ma│······························
trixARM -I/Users/lust/Documents/kiibohd-controller/Scan/MD2 -I/Users/lust/Documents/kiibohd-controller/Macro/PartialMap -I/Users/lust/Documents/kiibohd-controller/Output/pjrcUSB -I/Users/lust/Documents/kiibohd-controller/Debug/cli -I/Users/lust/Documents/kiibohd-controller/Debug/led -I/Users/lust/Documents/kiibohd-controller/Debug/p│······························
rint -I/Users/lust/Documents/kiibohd-controller/Debug/full -o CMakeFiles/kiibohd.elf.dir/Debug/print/print.c.o   -c ../Debug/print/print.c                                                                                                                                                                                                    │······························
/usr/local/bin/arm-none-eabi-gcc   -I.. -I.    -mcpu=cortex-m4 -DF_CPU=48000000 -D_mk20dx128vlf5_=1 -Os -mthumb -nostdlib -fdata-sections -ffunction-sections -fshort-wchar -fno-builtin -nostartfiles -Wall -ggdb3 -std=gnu11 -MMD -I/Users/lust/Documents/kiibohd-controller/Scan/ISSILed -I/Users/lust/Documents/kiibohd-controller/Scan/Ma│······························
trixARM -I/Users/lust/Documents/kiibohd-controller/Scan/MD2 -I/Users/lust/Documents/kiibohd-controller/Macro/PartialMap -I/Users/lust/Documents/kiibohd-controller/Output/pjrcUSB -I/Users/lust/Documents/kiibohd-controller/Debug/cli -I/Users/lust/Documents/kiibohd-controller/Debug/led -I/Users/lust/Documents/kiibohd-controller/Debug/p│······························
rint -I/Users/lust/Documents/kiibohd-controller/Debug/full -o CMakeFiles/kiibohd.elf.dir/Debug/led/led.c.o   -c ../Debug/led/led.c                                                                                                                                                                                                            │······························
Linking C executable kiibohd.elf                                                                                                                                                                                                                                                                                                              │······························
/usr/local/Cellar/cmake/3.0.0/bin/cmake -E cmake_link_script CMakeFiles/kiibohd.elf.dir/link.txt --verbose=1                                                                                                                                                                                                                                  │······························
/usr/local/bin/arm-none-eabi-gcc    -Wl,-headerpad_max_install_names   -mthumb -nostdlib -fdata-sections -ffunction-sections -fshort-wchar -fno-builtin -nostartfiles -Wl,-Map=link.map,--cref -Wl,--gc-sections -Wl,--no-wchar-size-warning -T/Users/lust/Documents/kiibohd-controller/Lib/mk20dx128vlf5.ld CMakeFiles/kiibohd.elf.dir/main.c│······························
.o CMakeFiles/kiibohd.elf.dir/Lib/mk20dx.c.o CMakeFiles/kiibohd.elf.dir/Lib/delay.c.o CMakeFiles/kiibohd.elf.dir/Scan/ISSILed/led_scan.c.o CMakeFiles/kiibohd.elf.dir/Scan/MatrixARM/matrix_scan.c.o CMakeFiles/kiibohd.elf.dir/Scan/MD2/scan_loop.c.o CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.o CMakeFiles/kiibohd.elf.dir/Output│······························
/pjrcUSB/output_com.c.o CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_dev.c.o CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_keyboard.c.o CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_mem.c.o CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_serial.c.o CMakeFiles/│······························
kiibohd.elf.dir/Debug/cli/cli.c.o CMakeFiles/kiibohd.elf.dir/Debug/led/led.c.o CMakeFiles/kiibohd.elf.dir/Debug/print/print.c.o  -o kiibohd.elf                                                                                                                                                                                               │······························
CMakeFiles/kiibohd.elf.dir/Scan/MD2/scan_loop.c.o: In function `I2C_BufferLen':                                                                                                                                   [... cut ...]
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2                                                                                                                                                                                                                                                                                         │······························
make: *** [all] Error 2                                                   
```

I am just hoping that this isn't actually ready yet and I should actually be using MD1. (which I have been until now)

haata, 2015-06-14:
MD2 is/was supposed to come right after the Infinity keyboard.
Unfortunately, stuff happened and that keyboard was mostly shelved.
The main interesting parts of MD2 became MDErgo1 (Infinity ErgoDox).

I may use MD2 for a future keyboard though.

kaansoral, 2015-06-14:
Wow, very curious :)

How was MD2 like?

closed 2016-01-13 by haata