kiibohd/controller #54: COM port for Windows?

unphased, 2015-06-14:
Sorry about starting so many damn issues!

I figured it would be nice if I could use PuTTY on windows to connect to the USB serial in order to use the debugger interface. 

But it seems like the drivers arent autoinstalled. It tries to Windows Update some drivers, they fail, and it appears to "fall back" to just being a keyboard, which is fine, but I've got no COM port.

unphased, 2015-06-14:
Ah, never mind. I found in the Wiki, `Teensy Virtual Serial Port`. I'll be back if I still can't figure it out.

closed 2015-06-14 by unphased

haata, 2015-06-14:
Windows, being a pain in the ass as usual :P

unphased, 2015-06-14:
Yeah i got the driver, which is good, since plugging it in is no longer a perplexing ordeal for windows anymore, but PuTTY flat out can't open the port even though Device Manager says it's the right port. Oh well. At least I have my mappings working and I can use this keyboard as my main keyboard now.

haata, 2015-06-14:
Windows tends to "lock" things like serial ports if another application
already has it open iirc. Dunno how to clear it, but a reboot usually did
it.

unphased, 2015-06-14:
I take back what I said about windows not being confused. It still tries to install drivers for 2 minutes after plugging the thing in and refuses to enable keyboard capability until it's done failing to fetch drivers. 

I gave up on serial, maybe I'll try it again after reboot. Goddammit windows.