kiibohd/controller #56: Fail to Load, Shows no errors.

Physonium, 2015-06-18:
I compiled and attempted to load onto a Teensy 3.1, and it failed to do so without a meaningful error message, at least that I noted.

Console Output:
http://pastebin.com/XJ1stiCq

I can supply the rest of my configs, I just don't know what's needed. I don't think I'm doing anything wrong, at least according to the documentation (to the best of my knowledge).

haata, 2015-06-18:
Hmm, a quick look at the logs seems like you are doing the right thing.

What platform are you on? Mac/Linux/Windows/etc

Physonium, 2015-06-18:
Should have specified, Linux (arch), everything up to date.

Physonium, 2015-06-18:
Problem resolved. Needed to run "make" after my cmake.
mkdir build; cd build; cmake ..; make; ./load 
Proper command to run to build once you have everything configured.

closed 2015-06-18 by Physonium