kiibohd/controller #57: Volume keybindings not working in Windows

flipjs, 2015-06-29:
I just brought my infinity keyboard at work which I am using Windows. The volume up, down and mute all work on my macbook pro (yosemite) but not on windows. No response or anything. I haven't checked the play/prev/play keys yet, but will update/post more on it.

The firmware is compiled and flashed using Mac. Any ideas on how to solve this? I plan to use this keyboard at work where I only use Windows (and Linux VMs). Thanks.

haata, 2015-06-29:
So, where you compile the firmware does not matter.

So there are two different keys for Volume Up/Down

U["VolumeUp"] same as U[0x80]
U["VolumeDown"] same as U[0x81]

as well as

CON["VolumeUp"] same as CON[0x0E9]
CON["VolumeDown"] same as CON[0x0EA]

Consumer Control Codes vs. USB Keyboard Codes

You can find both of them in the KLL spec.
https://www.overleaf.com/read/zzqbdwqjfwwf

Depending you may be able to send both at the same time.

U["A"] : U["VolumeUp"], CON["VolumeUp"];

Hopefully it doesn't do it twice on a Mac. Basically this is an OS issue because the keyboard has no way of knowing it's connected to a Mac or Windows machine.

flipjs, 2015-06-29:
I will try it tomorrow and let you know. Thanks Haata.

dumle29, 2015-07-09:
@haata Any way to do this through the configurator? Perheps use the CON ones as standard?

haata, 2015-07-09:
I can have both volume up/down added to the configurator. But if the CONS Volume up works everywhere I don't mind switching to it :D

flipjs, 2015-07-09:
Both the cons and normal bindings dont work on Windows, but work fine on my mac. Also, media keybindings dont work as well on Windows. I have asked this on Massdrop discussion if they use their infinity on windows, but no one bothered to answer.

flipjs, 2015-07-09:
I think you already have volume up/down in online configurator? That's how I started my keymap and edit from there and use this controller software to recompile and generate the bin from terminal.

haata, 2015-08-19:
Finally got Windows media keys fixed. Also included a bunch of random USB fixes (enumeration should be faster, and less problems with the virtual serial port). For Windows you must use CONS"VolumeUp" instead of U"VolumeUp" (both work for Linux and Mac).

20b62afb9a05a64b1f15c6329866600913775eea

Once I get a chance to verify that everything is good on the IC60 (Infinity) I'll push the update to the configurator.
http://configurator.input.club/latest/ is updated with the latest every couple hours if you're impatient.

closed 2015-08-19 by haata