kiibohd/controller #58: NKRO/Boot mode capabilities not working on teensy 3.1?

schizoidpig, 2015-07-31:
So I wired a keyboard with a teensy 3.1, and everything worked exactly as I wanted to, except I could not use my keyboard in my grub bootloader when I start up my computer. So I configured f9 and f10 to set NKRO and Boot respectively in my function layer. However, this seems to have no effect and I still cannot use my keyboard in grub. Also I used this link https://www.microsoft.com/appliedsciences/content/projects/KeyboardGhostingDemo.aspx to verify that my keyboard is always in NKRO mode.

Here is the code in my kll file, accessed via layerShift ( 1 );
U"F9" : kbdProtocolNKRO();
U"F10" : kbdProtocolBoot();

I'm using arch linux 64bit.

dumle29, 2015-08-02:
If you connect to the keyboard using a serial monitor (On the MD1 you have to jumper the two center header pins, I don't know if it's different on the other versions)

hit enter, then type help and hit enter.
You should get a list of commands. Run kbdProtocol. 1 = NKRO 0 = BIOS

schizoidpig, 2015-08-03:
Okay I used "screen /dev/ttyACM0" in my terminal to connect to the keyboard. kbdProtocol returns 1.

dumle29, 2015-08-03:
What OS are you running?

schizoidpig, 2015-08-03:
Arch linux 64bit

dumle29, 2015-08-03:
what does the command return if you set it to BIOS mode?

schizoidpig, 2015-08-03:
Well, after using fn +f10, which is set to kbdProtocolBoot(), there is no change.

dumle29, 2015-08-03:
odd

Streetwalrus, 2015-10-06:
I can confirm this issue on the same setup, using a Teensy 3.1 and the keyboard always reports being in NKRO mode even if I map a key to switch to boot mode and press it. Keyboard doesn't work in the BIOS/bootloader as a result.

haata, 2015-10-07:
I shall look at this in the next couple of days.

Streetwalrus, 2015-10-07:
I've just tried again, connecting the keyboard directly to the PC without a hub allows me to use it in the UEFI menu and in rEFInd, probably because my UEFI has extended HID support (you can use the mouse to browse the menu). That said the keyboard is still stuck in NKRO mode and I can't "hold DEL to enter setup", though that might be due to the firmware taking a little while to init.

haata, 2019-02-18:
This should working now after a bunch of USB fixes in the past few months.

closed 2019-02-18 by haata