kiibohd/controller #60: Unnecessary scrollbars added to windows in OS X Yosemite

dalkire, 2015-09-05:
When using the Infinity keyboard, the firmware will add unnecessary scrollbars to textEdit and iTerm2 windows for example. When I press the button to flash the firmware, the scrollbars disappear, but as soon as the firmware is loaded, they show up again. Possibly related to joystick and mouse descriptors.

johnhamelink, 2016-01-10:
This happens in El Capitan also.

haata, 2016-01-13:
I'm going to try to see if I can reproduce this, haven't been super successful so far...

tmk, 2016-01-14:
I guess the keyboard has mouse function? If so this will help.

https://github.com/tmk/tmk_keyboard/issues/74

haata, 2016-01-14:
> tmk/tmk_keyboard#74 <https://github.com/tmk/tmk_keyboard/issues/74>

Thanks hasu!

haata, 2019-02-18:
Should be resolved now.

closed 2019-02-18 by haata