kiibohd/controller #61: readLEDs reporting incorrect values

Streetwalrus, 2015-10-07:
I've wired up and configured my keyboard with a Teensy 3.1, and I was hoping I could come up with a hack to enable a numpad layer (it's a Truly Ergonomic, was a hell to mod but the original controller was fried) when numlock is on.
Still haven't wired the status LEDs but it looks like the controller is not updating the LED register value properly as readLEDs always returns the value the LEDs had when plugging the keyboard in.
I have a Logitech keyboard hooked up to the same machine and the LEDs behave properly on that. I'm running Arch 64 bit and I'm affected by #58. Happens both in X11 and in the virtual console.

haata, 2015-10-07:
Interesting.
These were working, though I might have broken this when I fixed a bunch of USB issues in the last few months.

Streetwalrus, 2015-10-07:
I've started investigating the issue, enabled all debug in usb_dev.c and I get the following output when any LED changes to any state (toggling them with another keyboard to make sure I only get LED packets).
```
CONTROL - pid:0xD, count:0x80034 - bmRequestType:0x21, bRequest:0x9, wValue:0x201, wIndex:0x1, len:2
WARNING - SET_REPORT - 0x201 - 0x1
STALL
CONTROL - pid:0xD, count:0x80034 - bmRequestType:0x21, bRequest:0x9, wValue:0x200, wIndex:0x0, len:1
WARNING - SET_REPORT - 0x200 - 0x0
STALL
```
So it's indeed trying to update the LEDs but not receiving the proper data from the PC. Since it happens on both Linux and Windows I suppose it might be a descriptor issue since the possibility of an OS bug is already cleared.

haata, 2015-10-09:
The USB portion seems to work correctly for me.

0x1 - Numlock
0x2 - Capslock
0x4 - Scrolllock

readLEDs only reads the current status. It doesn't automatically trigger when a change occurs (even though the state is recorded).

I do have some draft spec details on how to use USB Leds as triggers to combinations, but I won't have time to get to do the implementation until the first set of ErgoDox issues are fixed.

The easiest thing to do for now would be just to modify the led handling code here: https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/arm/usb_dev.c#L426

Now, this may be a 6KRO issue (looking at that next).

Streetwalrus, 2015-10-09:
Odd, it doesn't work for me. readLEDs always reports 0 or 1 regardless of numlock and capslock. Seems to change at random when I change the LED states.

haata, 2015-10-09:
Hmm, I'm getting random status as well now.
Time to pull out the USB analyzer :D

tmk, 2015-10-09:
I think Set report should read "data" for led status instead of wdata.

Streetwalrus, 2015-10-09:
Well I just checked with usbmon, the computer is indeed sending bogus data to the keyboard and not always, while my Logitech keyboard is getting proper reports.

haata, 2015-10-10:
I've been messing with the code. Something larger is wrong, should have a fix soon.

haata, 2015-10-10:
Fixed. Tested under both 6KRO and NKRO modes and seems to be working.
f4da2560c43f8182aa0fd8270fe175148d084b66

Let me know if you still have issues.

closed 2015-10-10 by haata

Streetwalrus, 2015-10-10:
Seems to work fine here, thanks a lot. Time to grab the soldering iron again.

haata, 2015-10-10:
There isn't a facility (yet) unfortunately. But I'll be adding somewhere around here: https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/arm/usb_dev.c#L692

Streetwalrus, 2015-10-10:
Thanks, I'll see what I can do with that.