kiibohd/controller #62: KLL Layout error during build

adamsaj, 2015-10-10:
I'm trying to build for the Infinity ErgoDox according to the "How To Load Layouts to Keyboard" wiki page, and I'm getting the following error message using the ergodox.bash script:

`funcparserlib.lexer.LexerError: cannot tokenize data: 88,16: "U"PERIOD" : U"V";"`

It displays this message for whatever trigger:response pair is on the bottom of my layout file.

haata, 2015-10-10:
Hmm, seems to be working on my end. Try making sure you haven't forgotten a semicolon.

If you still have an issue, please attach the entire kll file here.
Thanks.

adamsaj, 2015-10-10:
I've put the kll files on github at adamsaj/keyboard_layouts. 

The only other thing I've changed is adding the layouts to the build script:
```
# This is the default layer of the keyboard
# NOTE: To combine kll files into a single layout, separate them by spaces
# e.g.  DefaultMap="mylayout mylayoutmod"
DefaultMap="mdergo_base lcdFuncMap"

# This is where you set the additional layers
# NOTE: Indexing starts at 1
# NOTE: Each new layer is another array entry
# e.g.  PartialMaps[1]="layer1 layer1mod"
#       PartialMaps[2]="layer2"
#       PartialMaps[3]="layer3"
PartialMaps[1]="mdergo_numpad"
PartialMaps[2]="mdergo_arrows"
PartialMaps[3]="mdergo_qwerty"
PartialMaps[4]="mdergo_boot"
```

haata, 2015-10-13:
See my pull request https://github.com/adamsaj/keyboard_layouts/pull/3

adamsaj, 2015-10-19:
Yikes, I thought I had a better handle on the syntax. Thanks for your help.

closed 2015-10-19 by adamsaj