kiibohd/controller #64: Error building infinity.bash under OSX 10.11 (El Capitan)

nemtsov, 2015-10-13:
There's an issue in OSX 10.11 (El Capitan) where the mechanical keyboards (infinity keyboard in particular) is recognized (in System Information) but the keyboard does not work. 

It is also reported here for ducky keyboards (http://apple.stackexchange.com/questions/208576/mechanical-keyboard-stopped-working-after-upgrading-to-el-capitan/209174#209174).

So, hoping that updating to the latest firmware will solve the problem, I attempted to build `controller` and got the errors below. I appreciate any help you can provide on this, as typing without my keyboard sucks. (I also tried the VM to avoid build ussues, but since the keyboard is in a weird state the VM does not recognize it.)

Thanks!

```
➜  Keyboards git:(master) ./infinity.bash
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx128vlf5
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- Detected Scan Module Source Files:
Scan/MatrixARM/matrix_scan.c;Scan/MD1/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c
-- Detected Output Module Source Files:
Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/local/bin/git (found version "2.2.1")
-- Found Ctags: /usr/bin/ctags
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/delay.c and /Users/yuriy/wd/kiibohd-controller/Lib/delay.c: micros (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/delay.c and /Users/yuriy/wd/kiibohd-controller/Lib/delay.c: delay (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.c and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.c: fault_isr (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/delay.h and /Users/yuriy/wd/kiibohd-controller/Lib/delay.h: _delay_ms (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/delay.h and /Users/yuriy/wd/kiibohd-controller/Lib/delay.h: _delay_us (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/delay.h and /Users/yuriy/wd/kiibohd-controller/Lib/delay.h: millis (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/delay.h and /Users/yuriy/wd/kiibohd-controller/Lib/delay.h: delayMicroseconds (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/Interrupts.h and /Users/yuriy/wd/kiibohd-controller/Lib/Interrupts.h: cli (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/Interrupts.h and /Users/yuriy/wd/kiibohd-controller/Lib/Interrupts.h: sei (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: PORT_PCR_IRQC (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: PORT_PCR_MUX (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SIM_SOPT2_CLKOUTSEL (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SIM_CLKDIV1_OUTDIV1 (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SIM_CLKDIV1_OUTDIV2 (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SIM_CLKDIV1_OUTDIV4 (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SIM_CLKDIV2_USBDIV (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SMC_PMCTRL_RUNM (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SMC_PMCTRL_STOPM (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: SMC_VLLSCTRL_VLLSM (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: PMC_LVDSC1_LVDV (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: PMC_LVDSC2_LVWV (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: MCM_CR_SRAMLAP (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: MCM_CR_SRAMUAP (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: AXBS_CRS_PARK (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_CEEI_CEEI (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_SEEI_SEEI (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_CERQ_CERQ (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_SERQ_SERQ (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_CDNE_CDNE (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_SSRT_SSRT (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_CERR_CERR (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_CINT_CINT (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_DCHPRI_CHPRI (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_TCD_ATTR_SMOD (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_TCD_ATTR_SSIZE (Warning only)
Duplicate entry in files /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h and /Users/yuriy/wd/kiibohd-controller/Lib/mk20dx.h: DMA_TCD_ATTR_DMOD (Warning only)
[... 196 lines cut ...]
  File "/Users/yuriy/wd/kiibohd-controller/kll/funcparserlib/parser.py", line 200, in _shift
    (v, s2) = self.run(tokens, s)
  File "/Users/yuriy/wd/kiibohd-controller/kll/funcparserlib/parser.py", line 200, in _shift
    (v, s2) = self.run(tokens, s)
  File "/Users/yuriy/wd/kiibohd-controller/kll/funcparserlib/parser.py", line 158, in _add
    (v1, s2) = self.run(tokens, s)
  File "/Users/yuriy/wd/kiibohd-controller/kll/funcparserlib/parser.py", line 158, in _add
    (v1, s2) = self.run(tokens, s)
  File "/Users/yuriy/wd/kiibohd-controller/kll/funcparserlib/parser.py", line 182, in _or
    return other.run(tokens, State(s.pos, e.state.max))
  File "/Users/yuriy/wd/kiibohd-controller/kll/funcparserlib/parser.py", line 200, in _shift
    (v, s2) = self.run(tokens, s)
  File "/Users/yuriy/wd/kiibohd-controller/kll/funcparserlib/parser.py", line 201, in _shift
    return f(v), s2
  File "../../kll/kll.py", line 235, in make_usbCode
    return make_hidCode( 'USBCode', token )
  File "../../kll/kll.py", line 205, in make_hidCode
    hidCode = kll_hid_lookup_dictionary[ type ][ token[1:-1].upper() ][1]
KeyError: '{'
make[2]: *** [generatedKeymap.h] Error 1
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [all] Error 2
Firmware has been compiled into: 'IC60'
/Users/yuriy/wd/kiibohd-controller/Keyboards
```

haata, 2015-10-13:
Oops, really sorry about that. I fixed a symbolic name in KLL (has been wrong for a really long time) and forgot to update the hhkbpro2 kll file.

Do a git pull in the kll directory and it should fix the issue.

https://github.com/kiibohd/kll/commit/c3ec73860d8e163b61130a7d893d293144b88324

Let me know if you have any other issues.

closed 2015-10-13 by haata

nemtsov, 2015-10-13:
That fixed the error; thanks! (let's hope it fixes the usb issue too. will report here for future reference)