kiibohd/controller #66: Layer toggles across ergodox can get stuck in the wrong state

r4v5, 2015-10-19:
With the "stock" ergodox layout, pressing the two f1 keys at the same time (across both keyboards) and then releasing them in order will lock the keyboard in Layer 1 until the process is repeated.

Steps to reproduce:
* Press down left half's f1 key
* Press down right half's f1 key
* Release left half's f1 key
* Release right half's f1 key
* Observe the keyboard thinks it's back on L1 now.

I'd expect f1 to be cleared when both keys are up, and if you release the second key before the first, this does indeed happen. Also, when using a layout that has multiple f1 keys on the same half-keyboard, this issue does not seem to manifest for those two keys, so it seems that operating across the split is required.

haata, 2015-10-19:
Yep, definitely a bug.
I've reproduced it here. I think the issue is that the shift current XOR's the current state. But since it's XOR'd twice it's getting confused.

I should be able to fix it in the next few days, but if someone gets impatient, this is where the code is:
https://github.com/kiibohd/controller/blob/master/Macro/PartialMap/macro.c#L171

dumle29, 2015-12-17:
Just wanted to chime in. Happens on my MD1 as well, with this layout [image]

ntietz, 2016-10-09:
I've also run into this on my Infinity Ergodox. I am interested in contributing a fix. @haata can you point me in the right direction on this?

haata, 2016-10-09:
Basically, in this function (and the functions calling it) https://github.com/kiibohd/controller/blob/master/Macro/PartialMap/macro.c#L139

I believe what's happening is I'm using an XOR instead of just a set/unset. This means if you pressed two Shift keys at different times (now the layer is temporarily unset). But release them at the same time (I think this is what's happening), it will only unset once.

It should be possible to fix this by strengthening the logic a bit more. Instead of blindly setting every time the function is called.

ntietz, 2016-10-10:
I've submitted a PR that should resolve the issue, assuming the CI checks are successful. On my Infinity Ergodox, it fixes things.

ntietz, 2016-10-14:
@haata I believe this has been resolved, can we close the issue?

haata, 2016-10-14:
Sure, still need to update the configurator, @jbondeson will take care of that.

closed 2016-10-14 by haata

jbondeson, 2016-10-14:
@haata Configurator has been updated.