kiibohd/controller #67: LED brightness rollover

ibaranov-cp, 2015-10-22:
Hello,

Seems like you had planned for 255 levels of brightness for backlight, but hitting the LED+ key repeatedly gets about 4 levels until it rolls over and sets to lowest level again.

Any chance to avoid this behaviour?

haata, 2015-10-22:
So, it's not 4 levels, it just updates really really quickly right now (updates every 30 ms). I am planning a lot more sophisticated led control in the future but this is a ways out.

This is the KLL capability used to control the backlighting: https://github.com/kiibohd/controller/blob/master/Scan/ISSILed/capabilities.kll#L23
https://github.com/kiibohd/controller/blob/master/Scan/ISSILed/led_scan.c#L668

This is the default settings in the UI configurator: https://github.com/kiibohd/KiiConf/blob/master/js/defaults.js#L304

Some easy ways to get around this immediately:
 * Used fixed level control
 * Slow down the update rate (increase 1-5 levels per 30 ms)

I'm definitely open to changing the default behaviour (I was just trying to give a mini demo of what sorta of capabilities this led driver chip has).

erlandsona, 2016-05-01:
Well this definitely answers my question of the two hands updating out of sync a bit... LOL.
I just got my Ergodox put together with LED's this past weekend.

I'm wondering if there's a way to create the raindrop effect that the MK Disco keyboard has... EG: When I press a key the led for that key light's up and fades away in a few seconds.

Thoughts on how I could accomplish something like this? I'm not a very savvy C/Python programmer as I work primarily with Web Tech, so Ruby & JS (and a little Haskell) but I can pick up simple things in C / Py pretty quick if need be.

F1LT3R, 2016-05-27:
+1 To that idea @erlandsona!

F1LT3R, 2016-05-27:
@ibaranov-cp where is the LED + 1 key? My backlight is too bright to use at night, I would love it if I could turn them down.

ibaranov-cp, 2016-05-30:
Take a look here, you will see the LED- LED+ LED OFF buttons on the right.
http://configurator.input.club/latest/

In the code itself, it is here (as haata mentioned):
https://github.com/kiibohd/KiiConf/blob/master/js/defaults.js#L304

haata, 2019-02-18:
LED behaviour has been changed to not rollover since the integration of LED control to KLL.

closed 2019-02-18 by haata