kiibohd/controller #68: This firmware causes Windows 10 installation to fail / freeze

kaansoral, 2015-10-24:
I don't use the infinity firmware anymore yet the keyboard I built for my brother had the infinity firmware, for a full day we tried to figure out why the Win10 installation was failing, tested 100's of bios settings, removed components, disabled/enabled raid, pretty much eliminated anything

At best, the Windows 10 setup was freezing at "Just a Moment ..."

Anyway, long story short, removing the keyboard fixes the issue (instantaneously), the firmware creates a deadlock somehow, if the keyboard is attached, the installation freezes forever at that step

The firmware is a 2-3 months old one, yet it's likely that others will experience this issue too, I hope I saved someone from suffering the same issue, if you are having weird issues, test things again without the keyboard

(After the issue my brother mentioned other anomalies like these, but I'm guessing they were regular bugs, like the shift modifier getting stuck etc. and not as critical as this one, yet anything is possible)

dragon788, 2016-01-09:
I'm thinking that its likely that the firmware isn't properly reporting 6KRO instead of NKRO when it gets to the Windows installer because it sees that the BIOS has handed it off. I'm curious whether this is still an issue with current ones.

haata, 2016-01-09:
So, BIOSs often have settings called EHCI and XHCI handoff. I've read that XHCI hand-off is required for things to work in general (USB 3.0 stuff).
But EHCI, I'd recommend disabling as that keeps the devices enabled and hands off the settings to the OS. In the case of NKRO keyboards, you want the OS to reset the device and re-enumerate to go into the proper mode.