kiibohd/controller #69: Certain characters won't output.

danelowe, 2015-10-26:
Brand new ErgoDox infinity. After building, I notice that only certain characters seem to be working out of the box and others aren't. OSX El Capitan.

### Working output
`a-Z`, `0-9`, ` !-)` (those usually got by shift+number row), `backspace`, `space`, `enter`, `tab`, `shift`, `FLASH, f1`, `gui, alt, ctrl`

### Not being output
`-\'[],.<>/=+?{}_|"` and arrow keys.

I figured it just needed flashing, so set up a few layouts using the online configurator and flashed both halves individually. 

Things I noticed:

* If I map the non-working characters to a previously working key, the character will not output when pressed.
* If I map a working character to a previously not-working key, the character will output when pressed.
* If I press a non-working character key in quick succession after a working character key, it sometimes outputs the non-working character, and then repeatedly outputs the non-working character until another working character key is pressed.

I have no idea where to start to debug this further.

danelowe, 2015-11-14:
Had a play with `screen`, and `outputDebug` is showing something. I assume the codes that it is sending for the non working keys are not recognised by the OS? In which case it could be fixed by mapping those codes to others in the KLL? 

I just have no idea how to read the codes output from `screen` e.g pressing what should be the `\` key. 

```
DEBUG - NKRO USB: 00 000000000020 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000000000000 0000000000000000000000000000 00 000000000000
```

How would I go about using this information to map the keys correctly?

danelowe, 2015-11-14:
Just tested on a windows laptop, and it all works correctly. Just doesn't work on my macbook.

haata, 2015-11-14:
(I'm currently on a cycling trip across NZ, internet has been iffy, so it may take me a while to respond)

Hmm, that's interesting it doesn't work on a macbook.
It really does seem like the keypress is being sent.

What locale is your keyboard (US by default)?
Do you have any re-mapping software running?

I hadn't tested the macro keys on a Mac (I'd hoped they would work), but it may be that Apple wants to be a special princess (yet again). Try hold shift before pressing any of the keys. If this works, Mac OSX is incapable of processing a Shift+1 (to be !) even if they are pressed at the exact same time (which is totally valid).

If this really is the case, I'll have to think about how to handle this because sure, it's an easy fix if there are only two keys in the combo. But how about 20? You start to get time delays that shouldn't exist in macros.

danelowe, 2015-11-14:
Oh man, it was just software I had installed (USB Overdrive) to re-map the caps lock key to escape on the internal keyboard! I was so used the re-mapping that I had forgot I had the software installed. 

Thanks so much for the reminder I so obviously needed. 

Do you mean up/down NZ? Sounds awesome. I've left my ergodox alone for 3 weeks without it working, but was desperate to get it working in the next few days because I'm heading back home to NZ for a month and my kinesis would be too big to fit in my bags.

closed 2015-11-14 by danelowe

haata, 2015-11-14:
Haha, nice.

Yeah, I started in Auckland on Nov 1st. Currently in Wellington. I'm gonna
try to go all the way south if my legs (and behind) allow me to.
I fly back to SF from Auckland on Dec 1st.

I'm finally starting to feel recharged, which will mean more work on
keyboards.