kiibohd/controller #74: compiler error in ergodox.bash while accesing layer keys with from upper layers in kll

20lives, 2015-11-18:
[ 26%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.o
In file included from ../../Macro/PartialMap/macro.c:30:0:
./generatedKeymap.h:242:27: error: ‘KEY_FUN2’ undeclared here (not in a function)
Guide_RM( 92 ) = { 1, 14, KEY_FUN2, 0 };
                          ^
make[2]: *** [CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.o] Error 1
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [all] Error 2
Error in make. Exiting...

adamsaj, 2015-11-19:
I have also observed this issue using both Function and Lock keys.

20lives, 2015-11-19:
note that the kiiconf can do that.

tmk, 2015-11-19:
I'm not well at KLL but my guess is:
U"Function2" or U"Fun2" is virtual key(not real USB keycode) and you have to map action(result macro) to the key somewhere. Usually those 'Function' keys are mapped in stdFuncMap.kll or lcdFuncMap.kll. You will get the error unless those keys are defined.

adamsaj, 2015-11-19:
@tmk, You are correct. I was able to get the firmware to compile by adding the stdFuncMap as a modification to the partial map like this:

`PartialMaps[1]="mdergo_numpad stdFuncMap"`

I'm a little curious why this isn't necessary for the defaultMap, which can call Function keys just fine while being defined like this:

`DefaultMap="mdergo_base lcdFuncMap"`

haata, 2015-11-20:
(Sorry, internet has been really shoddy while I've been travelling in NZ,
and thanks hasu!)

The reason why it's necessary to add stdFuncMap or lcdFuncMap is because
kll needs to understand the order in which the replacements are done.
This is because U"Fun" is not a real USB code (as hasu mentioned).

The reason why you need to add either stdFuncMap or lcdFuncMap to Default
and each Paritial map is because internally the keymaps are not related.
Both the Default and Partial maps start with the BaseMap and then any
layouts specified override what was specified in the BaseMap.
You can't just add the lcdFuncMap to the BaseMap because the symbolic names
(Func1, Lock1, etc.) would be lost (as Func1 to layerShift(1) is an
arbitrary mapping).

Hope this helps. I'll be back in the US in December (and should be ready to
add some new features :D).

20lives, 2015-11-21:
thank you all for the help.
another error occur when trying to sequence or combine layes commands.
anny suggestions what to do here?

and by the way HaaTa, combination and squence in the same key isn't possible also.

tthayer, 2016-01-22:
@adamsaj Thanks for the tips. I was running into this problem and was able to compile cleanly finally. Working great!

adamsaj, 2016-02-06:
A note to anyone who stumbles on this trying to fix the same problem:

You want to use `lcdFuncMap` rather than `stdFuncMap` for the Infinity Dox. This will update the LCDs when you push or pop layers from other layers.

paradox460, 2016-04-24:
To put it into even simpler terms, you need to put `lcdFuncMap` at the end of any `PartialMap` line.

Should be obvious, but took my 2 AM brain far too long to figure this simple fact out, so I thought I should write it down for other people's benefit.

haata, 2017-11-20:
I believe this has been resolved.

closed 2017-11-20 by haata