kiibohd/controller #75: keypress stuck on resume from sleep

cubiq, 2015-12-29:
OS: Arch Linux
Keyboard: WhiteFox

How to replicate: Put the computer to sleep. Wake the computer pressing the PC power button.

Result: The controller keeps repeating the first key you press when resuming from sleep. Pressing a second key stops the stuck key and the keyboard starts working as expected.

Not completely deterministic. I'd say it happens 50% of the times.

cubiq, 2016-03-11:
today "DEL" key got stuck and deleted a bunch on emails by accident!

this seems a pretty critical issue...

haata, 2016-03-11:
:/

Hmm, I had to do something similar for a capsense controller I did a long
time ago. Basically, ignoring any keypresses being pressed while turning on.

Question, do you have LEDs in that keyboard? i.e. are the LEDs on during
sleep mode? (trying to figure out what sort of state the keyboard is in).

cubiq, 2016-03-11:
no LEDs, I can add a couple just to check

haata, 2016-03-11:
I recently added code to deal with changes in USB power state (though I
haven't pushed them to the configurator yet, needed to get more runtime on
it). So I may be able to use these hooks to deal with this problem. Having
the leds will be a good way to visually indicate the modes.
Thanks.

cubiq, 2016-03-19:
okay so I installed LEDs. This is pretty weird and don't know if it is intended behavior, but anyway backlight stays lit when the PC goes to sleep. Also I can keep altering LED brightness with the key combo I programmed (FN2+page up/down) when the PC is asleep.

So LEDs don't seem to help with this issue.

intafon, 2017-11-15:
I think I actually ran into this issue on my KType yesterday a couple of times. Not sure if it was waking from sleep now, but I did have delete key stuck issues which required unplugging and plugging the keyboard.
(macbook pro, retina 15 inch, late 2013, keyboard connected via thunderbolt monitor usb port)

haata, 2017-11-20:
On macOS, I have an open bug related to HID IDLE timeouts (not defined on other OSs, usually).

Most of the issues should be fixed with https://github.com/kiibohd/controller/commit/a9e8619037bde4d27ca11303605a4957054bd872 though.

haata, 2017-11-22:
I've implemented this fix for HID IDLE timeouts https://github.com/kiibohd/controller/commit/cb040718ae8af8272e2bb30791f4186f4a25c74f

closed 2017-11-22 by haata

rwade628, 2019-02-13:
Not sure if I should open a new issue, but I'm still seeing this on my K-Type running v0.5.3

PandorasFox, 2019-04-05:
I'm also still seeing this on my Nightfox