kiibohd/controller #76: More than 254 TriggerMacro's bricks the device

kevinfrei, 2016-01-02:
If you have more than 254 TriggerMacro's in your KLL code there are a number of infinite loop's that occur in macro.c. I haven't figured out how to create a pull request for just that file, but I've fixed the issue in my fork of the repo (sorry, github n00b). You can see the diff (plus the change to the .kll file to support more than 255 of them) here: https://github.com/kevinfrei/controller/commit/ce013db72dd344afe9b5e909f0e87867df34931d

It would also be nice of KLL would warn that you need to update the stateWordSize, but I don't really do Python... (I've been avoiding it for 20 years...)

haata, 2016-01-02:
Yay! Someone trying to push the limit :D

To do a pull request for a specific file, create a branch from master. Then commit only that file to the branch and push it. During the pull request if you need to make changes you can just update that branch and it will automatically update the pull request.

I'll add myself a github issue on stateWordSize, I think it might even be possible to automatically detect (and update) the value from kll.
https://github.com/kiibohd/controller/issues/77

kevinfrei, 2016-01-03:
I think I managed to do it properly. Check out the pull request I created and let me know if I need to do anything differently. I would have added comments, but the comments wouldn't be useful once the code's been updated.

cryham, 2016-05-07:
Just tested and actually more than 255.
The next issue 77 is about making a fix for it, possibly working.

haata, 2016-05-18:
Seems fixed now thanks @cryham

closed 2016-05-18 by haata