kiibohd/controller #79: LED orientation error

ibaranov-cp, 2016-01-05:
Hello all,

I am suprised no one noticed this before, but for some reason LEDs in the lower half of the keyboard are reverse biased....

I installed all LEDs with short-leg to square pad, and the top half works as expected (dimming, brightning, etc) however 3 row and down are all dark. After some bug checking, noticed that a LED attached in reverse worked fine.

Hopefully there is a firwmare fix for this... Will look into it but pointers would be nice :)

haata, 2016-01-05:
Hi illia, yeah, because of the pwm matrix you can actually put the LEDs in
both ways, though there is an ideal way.
Ping me when I get back from ces and I'll try to get some instructions.

ibaranov-cp, 2016-01-06:
I disagree, unless there is something wrong with my board, the bottom half LEDs never light up under any condition, unless installed in reverse, and only then they work properly.

I will have to look into this more...

haata, 2016-01-09:
Got a breather from CES.

Basically, the ISSI led driver chip is a led matrix that takes advantage of
leds being diodes, driving both pins alternatively to do 144 led channels
using 18 pins.
The other thing that could be going on is that I didn't enable the correct
led locations and so it's only being turned on in the wrong direction.

(It's also possible the pcb is screwed up, but I want to make sure it's not
just a dumb software bug first).

Also, I've been playing with the LED driver a lot more for the KType
keyboard so I should have some more LED infrastructure ready pretty soon.
With the Ergodox I was able to sustain 83 fps (really helps that there are
only 38 leds to update over i2c).
In order to organize the RGB leds I wrote some cli tools to run through all
the led positions to make sure they were all working properly (might help
in this case as a quick test). I should have them ready for the Ergodox in
a couple weeks (currently hardcoded for the KType).

haata, 2016-01-13:
@ibaranov-cp also, do you have a list of the leds that you are having issues with?