kiibohd/controller #82: Cherry MX Lock switches

sigboe, 2016-01-07:
Please support [Cherry MX Lock](http://deskthority.net/wiki/Cherry_MX_Lock) switches. They are a linear switch with a latching mechanism, historically used for capslock.

If the firmware was aware of the latch, it could change the behavior if you told it the key had a latch switch. So that when you pressed down a latch switch the firmware would only send to the operating system that the key was momentarily pressed down, and when it released the firmware will tell the operating system that the key was yet again momentarily pressed down.

This way you could have a lock button like Capslock or a function lock, tell the firmware you want are using a lock switch and then have it function properly.

dragon788, 2016-01-09:
+1 as I've got some of these lock switches on order.

haata, 2016-01-09:
I'm not sure if the configurator currently supports this key, however, compiling yourself KLL definitely supports the latch versions of the lock keys.

Specifically, USB Codes 0x82, 0x83, and 0x84.
```
U"CapsLock" : U"CapsToggleLock";
U"NumLock" : U"NumToggleLock";
U"ScrollLock" U"ScrollToggleLock";
```

sigboe, 2016-01-09:
@dragon788 where did you order latch switches? I would like to order some my self.

@haata I didn't know this was a feature already. Does it only work for CapsLock, NumLock and ScrollLock or functionlock keys as well? Maybe if the configurator would do the method you described using USB Codes like that for Caps, Num and Scroll, and then if you tell the configurator that you have a latch key on a functionlock key (and maybe other keys) it can do something like I described there or something better.

haata, 2016-01-09:
Yeah, it's only for those 3 keys (at least as far as the USB HID spec
goes). For other keys I'd have to put some code in to handle it properly.

Would you mind describing the sort of use cases you have in mind for the
lock switches? Personally I don't really like using those types of switches
(though I do think they are neat) so it's harder to think about how someone
would like it to work.

sigboe, 2016-01-09:
@haata Right now my main keyboard at home is the MD1 (I'm up for May shipment of the White Fox :smile: ) And one thing that is lacking from my secondary keyboard (Infinity ErgoDox) is the ability to at a glance understand where I am in the keyboard. I mean, there is no LED support or screens that give me some information to give me my bearings.

If I could solder on a lock switch for, lets say Caps Lock and Function 1 Lock, I could at a glance of the keyboard see that ~~cruise control~~ Caps Lock is on, or the Function 1 is locked without having any LED support. These are things that are on other keyboards very often signified with LEDs. On my laptop there is both a backlight and a separate green LED on the caps lock, and on my dreadful office keyboard (soon to be retired) there are Fn LEDs as well.

So the reason I want lock switches is to give a similar function to LEDs, when capslock is down, I know its on. When Function 1 lock is down.

When I think about it. I can just get a hold of some lock switches and then try to see if the USB code for CapsToggleLock works, and then use regular ƒ1 from the configurator on my Function Lock key.

haata, 2016-01-10:
> the MD1 (I'm up for May shipment of the White Fox [image: :smile:] ) And

I'm thinking it should actually just work. So give it a go, and see what
happens. The Function 1 Lock, just assign it as a function shift key and it
should behave exactly as you want it.

dragon788, 2016-01-10:
I ordered them from the ortholinear keyboard store olkb.co iirc. They are
still pending mostly because they are also fulfilling orders from a hugely
popular Massdrop sale of one of their keyboard models.

sigboe, 2016-01-10:
@dragon788 Wow, look at that price, after shipping the price will be 60USD for me for two switches (16USD before shipping). I'll be looking around first. Thanks anyway.

dragon788, 2016-01-10:
No problem Sigurd, the reason they are so high is that Cherry is no longer
making them, period, full stop, they are done. So unless Gateron or
Greetech or Kailh decides to make their own, there is a limited
availability of them. That's part of the reason I jumped on them now, just
to get enough for the next few keyboards I might want to make with physical
lock/latch keys.

sigboe, 2016-01-10:
@dragon788 They are only 8USD per switch on the website you liked to but the shipping was many many times the price for two switches. Maybe I am okay with mixing greetech or kailh with cherry... I dont know yet xD

sigboe, 2016-01-24:
I asked mechanicalkeyboards.co.id if they could stock it and they did.

https://www.mechanicalkeyboards.co.id/products/detail/cherry-mx-lock-switch-linear-locking-plate-mount

And it looks like they ship world wide without an outrageous cost.

sigboe, 2016-02-03:
@haata So I got my switches and installed one thus far. So as a test I try to change the json file and find my Lock button and change this

```json
        {
            "code": "0x82",
            "h": 4,
            "layers": {
                "0": {
                    "key": "LCK1",
                    "label": "LOCK-1"
                }
            },
            "w": 4,
            "x": 66,
            "y": 17
        },
```

Into this

```json
        {
            "code": "0x82",
            "h": 4,
            "layers": {
                "0": {
                    "key": "LCK1",
                    "label": "LOCK-1"
                }
            },
```

Then I import the json file into the web configurator and downlaod and flash. The button still does the layer change. Should I change more of the json or cant I do it this method?

haata, 2016-02-03:
I'd recommend doing changes using KLL for the most part. Using the json it
has to go through another conversion layer that may not have all the
symbols needed. Once we make sure it actually works, then we can look at
making the json/configurator work.

sigboe, 2016-02-03:
Thanks, I will look into how to compile KILL tomorrow after work. Now I need some sleep so I will wake up in time. :smiley:

ratratanx, 2017-04-22:
Hi! Anyone can give me or sell an MX Lock? Already posted on Deskthority, Geekhack and Reddit. No one is responding or wants to sell. Please email me ratratanx@gmail.com