kiibohd/controller #84: Problem Loading Layout to Infinity Ergodox

pce1991, 2016-01-22:
After doing the last step [here] (https://github.com/kiibohd/controller/wiki/How-To-Load-Layouts-to-Keyboard) the LED panel remains orange. This is the output from "sudo bash load"

Opening DFU USB device... ID 1c11:b007
Run-time device DFU version 0110
Found DFU: [1c11:b007] devnum=0, cfg=1, intf=0, alt=0, name="Kiibohd DFU"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
No valid DFU suffix signature
Warning: File has no DFU suffix
bytes_per_hash=514
Copying data from PC to DFU device
Starting download: [##################################################] finished!
unable to read DFU status

The board doesn't take any input, when I unplug it and plug it back in the panel is still orange, and pressing the button doesn't return it to normal.

haata, 2016-01-22:
So, the orange light staying on usually means the firmware isn't compiled properly for that chip (Gets into a bad state, then immediately goes back to flashing mode).

Quickly looking at those instructions, there's at least one missing step. For the Infinity Ergodox, you also need to change the `Chip="mk20dx128vlf5"` to `Chip="mk20dx256vlh7"` then re-run the script.
At least after this the orange light should stay off. 

I recommend using the ergodox.bash script instead for the Infinity Ergodox as it has most of the less obvious things taken care of.

WishCow, 2016-06-04:
I just ran into this, and it doesn't seem to be solved even after changing the chip line (I also deleted the build directory). 

Running the ergodox.bash script instead of the infinity.bash one produces all kinds of errors:

```
/home/wishcow/projects/controller/Scan/STLcd/lcd_scan.c: In function 'LCD_layerStackExact_capability':
/home/wishcow/projects/controller/Scan/STLcd/lcd_scan.c:383:5: error: 'STLcdNumber0_define' undeclared (first use in this function)
   { STLcdNumber0_define },
     ^~~~~~~~~~~~~~~~~~~
/home/wishcow/projects/controller/Scan/STLcd/lcd_scan.c:383:5: note: each undeclared identifier is reported only once for each function it appears in
/home/wishcow/projects/controller/Scan/STLcd/lcd_scan.c:384:5: error: 'STLcdNumber1_define' undeclared (first use in this function)
   { STLcdNumber1_define },
     ^~~~~~~~~~~~~~~~~~~
/home/wishcow/projects/controller/Scan/STLcd/lcd_scan.c:385:5: error: 'STLcdNumber2_define' undeclared (first use in this function)
...
```

haata, 2016-06-04:
It looks like you're missing a .kll file somewhere. Likely DefaultMap or lcdFuncMap. Any variable with a _define after it is generated with KLL.

WishCow, 2016-06-04:
Sorry my bad, I messed up the config. I managed to run it now, it created an ICED-R, ICED-L folder, but running `sudo ICED-R/load` produces the same result, keyboard stays in the flashing state (orange led).

WishCow, 2016-06-04:
Ok, using dfu-util -D instead of the load script worked. Not 100% sure if it was really the load script, or just me messing up again, sorry.

srang, 2016-10-29:
Yeah I also followed the config guide for the Infinity board [here](https://github.com/kiibohd/controller/wiki/How-To-Load-Layouts-to-Keyboard) but if I'm not mistaken that is for the Infinity _60%_ board not the ergodox. Maybe the resolution for this issue is to add a wiki page for customizing the Input Club Infinity Ergodox so as to make the distinction and add the tidbit above 

> it created an ICED-R, ICED-L ... using dfu-util -D instead of the load script worked. 

to help people out. I too ran into the `lcd_scan` problem and fixed it by leaving the `DefaultMap` as is and just adding a `PartialMap[0]`