kiibohd/controller #85: Change "latch" behavior

alexgleason, 2016-01-24:
Hello. Thanks for working on this. Coming from an Ergodox to an Ergodox Infinity I'm running into a problem with the latch functionality. Here's the behavior I'm wanting:

* As soon as the latch button is depressed, it should enter that layer
* If a key is pressed while the latch key is still depressed, it doesn't "unlatch" and remains in that layer allowing you to continue typing other keys as long as you remain holding the latch button
* When the latch key is released, if another key were typed while holding the latch button, the layer becomes "unlatched". Otherwise, the behavior is normal (the layer is unlatched when another key is typed)

This change shouldn't conflict with current uses of the latch key, but also allow me to continue using my specialized configuration.

So you can understand, this is what I previously had with my Ergodox:

[image]

The main thing to see is the ~L1 where Caps Lock normally is. When I'm using a text editor, I press and hold that button with my left pinky finger to allow me to traverse the editor with my free fingers using the keys on layer 2. Since I'll be tapping the arrow keys, I want the ability to stay in the layer without it locking. But since I'm typing and moving fast I also don't want to continually have to hit the button to lock and unlock it, so lock mode doesn't work for me either.

Can this be looked into? Thank you!

haata, 2016-01-24:
So, to confirm.

* Latch key that behaves as a function shift while held, but turns into a latch once released?

Let me think about it for a bit, but I think I agree with you about not affecting intended behaviour :D

alexgleason, 2016-01-25:
Exactly! More specifically:

* Latch key that behaves as a function shift while held (and a key is pressed while it's held), but turns into a latch once released (if a key wasn't pressed while it was held)

Thanks. :)

haata, 2016-01-26:
Cool, that sounds good.

The "if a key wasn't pressed while it was held" part is a bit tricky. However I have to implement just that for KLL 0.4 (which is what I'm working on right now).

I need to think about whether a modifier key is considered a key or not. If it is, do all types of modifiers affect this trigger. Do lock keys count? Other latch keys (latch chaining)?

alexgleason, 2016-01-26:
> I need to think about whether a modifier key is considered a key or not. If it is, do all types of modifiers affect this trigger. Do lock keys count? Other latch keys (latch chaining)?

I spent a very long time pondering this and stroking my mustache. I think I know the answer. Let the most recently pressed modifier take priority. The other ones don't matter and can safely be released.

Let's say you're holding a latch key (so it's acting as a shift). Then you press and release a different latch key. You're still holding the first latch key. They keyboard is now "latched" by the different latch key and you can safely release the first latch key. Now the keyboard behaves as though you had fully depressed/released the first latch key followed by the depress/release second latch key.

typorian, 2016-05-03:
I'm pretty sure the way alexgleason described it is the way this is implemented in TMK, which is quite neat as you can have lock layer keys on layers that you reach by holding down a momentary layer shift modifier (like the spacebar in spavefn).