kiibohd/controller #87: Port request Adafruit Feather M0

sigboe, 2016-01-26:
So my HTPC remote control was not functioning as intended for a few days, and I used my mech keyboard via a cable for the time. And I thought, isn't it great to use a mech on my HTPC (thats also made for gaming)

So Adafruit came to mind, that store has some great engineers and interactions with the customers. They designed this [Adafruit Feather M0 Bluetooth LE](https://www.adafruit.com/products/2995) which I think looks similar to the Teensy in what it has to offer. It uses an ATSAMD21G18 ARM Cortex M0. and is a few times the minimum specs of this firmware in regards to RAM and ROM, and has enough pins, Microusb, and charging circuits built in! The charging circuits automatically use USB power when available.

It has 20 GPIO pins, enough for a 60%. It has several more pins which I presume can be used to make a full size keyboard or back-lighting. The Teensy has an ARM Cortex M4, so it sound to me like the same processor family. It has the same clock speed as the Teensy 3.0 

I'm not actually a big fan of wireless keyboards, but they sure are convenient in many situations. Like HTPC, traveling with a tablet (I have an iOS, an Android, and an Intel china tablet which all have horrible on screen keyboards), some people bring their mechs with their laptops, and it would be splendid there as well.  

If this could become a reality, I would start making my dream 60% and try to source low profile caps. If this is out of the question, please close the ticket.

haata, 2016-01-26:
Hmm, that's an interesting module. And you're right, it does have the sort of specs required for KLL to work right (bluetooth stack will take more than USB, but still, lots of room left).

I agree, Bluetooth implementations thus far have been crap. I haven't seen anything that I'd say works well (and doesn't have a dongle).

So, reading more into it, it seems to use the Bluetooth chipsets from Nordic (which I had also been looking at) in addition to one of Atmel's newer ARM chips (only M0, compared to M4 in the Teensy 3s, which means it takes more instructions to do the same sorts of things, but you get lower power).

I fully expect to have to rewrite the bluetooth stack on the Nordic to make it work nicely, which is the main reason I haven't started yet.

Anyways, thanks for the suggestion. I'll keep this open for a while to remind me of what's available. I am thinking about bluetooth, but I'm not quite sure when I'm going to get to it.

sigboe, 2016-01-27:
@haata And Adafruit also carries batteries already terminated plug and play for this controller. 2500mAh is the biggest flat battery, and 6600mAh is the biggest thick battery. The last one would be usable for 13 hours constant max power draw. realistically it would last much longer on normal load. And longer still if you did some black magic and wrote good sleep functionality.

sfcgeorge, 2016-04-23:
I had exactly the same idea and would love to build a keyboard using it (I'm thinking an Atreus style). Re the Bluetooth stack, it says there's actually a keyboard mode already. I'm not exactly sure how that would work, but I think you just send keystrokes to it and it does the rest. So maybe a Bluetooth rewrite isn't necessary? https://learn.adafruit.com/adafruit-feather-m0-bluefruit-le/hidkeyboard

haata, 2016-04-23:
So, the keyboard would "work". Though I'm not sure if it would support NKRO, which can be an important feature when using KLL macro capabilities.

bam80, 2019-01-28:
Sorry for noise, is any progress regarding bluetooth keyboards? I'm new here so don't really know what direction to look at..

haata, 2019-01-28:
We've been doing some initial work for Bluetooth (though progress has been slower as we're still going through all of the Kira bugs and features).
Lately I've been leaning towards more of the nRF52 rather than the nRF51.

As for the X-1, I recently discovered it might be possible to make firmware for the X-1, though I'm not sure how easy it will be to load it onto the keyboard. It's also, not a Nordic based chip so it would be a lot more work.

bam80, 2019-01-28:
Thanks. Is there a place where that Bluetooth work can be tracked? I didn't find anything..
Any hope on some Bluetooth keyboards support in near future?