kiibohd/controller #88: Doesn't work with KVM

ec1oud, 2016-02-03:
I have an Aten CS-1764.  The keyboard is powered when plugged in (to the keyboard-specific USB port on the front) but I can't type anything.

haata, 2016-02-03:
Hmm, KVMs can be pretty difficult to get working (most, if not all of them don't follow the HID spec).
It may be necessary to force the keyboard into Boot mode to see if that helps.
I'm also pretty sure that the DFU bootloader will also not work through the KVM.

ec1oud, 2016-02-06:
No DFU doesn't work, but that's OK; I don't need to reprogram it so often.

jwcha, 2016-07-05:
o/

keyboard: WhiteFox, using the online configurator.
[WhiteFox-TrueFoxBlank-v14.zip](https://github.com/kiibohd/controller/files/348136/WhiteFox-TrueFoxBlank-v14.zip)

Attaching my config.

I hooked it up to the keyboard USB port on a IOGEAR GCS932UB (2 port DVI KVM switch) and it doesn't accept any inputs using NKRO. If I switch to 6kro, I can type and use my layers, but for some reason the ALT, SHIFT, and CTRL buttons do not work for me (they are mapped to the main layer).

Just wanted to log it here. For now I'll hook it up directly to one of my systems and use Synergy to have control over both machines.

haata, 2019-02-18:
I'd recommend putting the keyboard 6KRO mode (I've fixed a bunch of bugs since this was reported so everything should work fine now).

closed 2019-02-18 by haata