kiibohd/controller #89: spontaneous resetting periodically

ec1oud, 2016-02-06:
If I alter the backlight brightness, it's very easy to see when the firmware restarts: it goes back to default brightness.  (BTW the default 0xFF is crazy, like staring at a lamp.  I changed it to 0x03.)  It happens quite often, more than once an hour, and always on whichever half is the master (connected to the computer's USB), not the slave.  And it happens even when I'm not typing on the keyboard at all.

To reproduce: plug it in and turn the key-backlighting LEDs off (via whatever key stroke you have configured).  Set the keyboard aside and just wait.  After a while, the half that is plugged into the computer will go back to default brightness.

I'm using the latest controller firmware 8c858e4 with modified KLLs (customized keymap, nothing too strange, and custom initial LED brightness, but this happens also without customizing that).

ec1oud, 2016-02-06:
Also tested connecting to the serial port (ttyACM0 for me).  It disconnects when the firmware resets, naturally.

haata, 2016-02-06:
Interesting, would you mind checking dmesg after this happens and post the log (only need the stuff around that time)?
I have a sneaking suspicion about what it is...

ec1oud, 2016-02-07:
[752507.388544] usb 2-1.6: USB disconnect, device number 47
[752507.661809] cdc_acm 2-1.6:1.2: ttyACM0: USB ACM device
[752507.864917] usb 2-1.6: new full-speed USB device number 48 using ehci-pci
[752507.953373] input: Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.0/0003:1C11:B04D.04DB/input/input1253
[752508.005493] hid-generic 0003:1C11:B04D.04DB: input,hidraw5: USB HID v1.11 Keyboard [Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full] on usb-0000:00:1d.0-1.6/input0
[752508.007532] input: Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.1/0003:1C11:B04D.04DC/input/input1254
[752508.058717] hid-generic 0003:1C11:B04D.04DC: input,hidraw6: USB HID v1.11 Keyboard [Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full] on usb-0000:00:1d.0-1.6/input1
[752508.059176] cdc_acm 2-1.6:1.2: ttyACM0: USB ACM device
[752508.062427] input: Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.4/0003:1C11:B04D.04DD/input/input1255
[752508.062558] hid-generic 0003:1C11:B04D.04DD: input,hidraw7: USB HID v1.11 Mouse [Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full] on usb-0000:00:1d.0-1.6/input4
[752508.064152] input: Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.5/0003:1C11:B04D.04DE/input/input1256
[752508.064285] hid-generic 0003:1C11:B04D.04DE: input,hidraw8: USB HID v1.11 Joystick [Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full] on usb-0000:00:1d.0-1.6/input5
[752508.066335] input: Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.6/0003:1C11:B04D.04DF/input/input1257
[752508.118800] hid-generic 0003:1C11:B04D.04DF: input,hidraw9: USB HID v1.11 Device [Kiibohd Keyboard - MDErgo1 PartialMap pjrcUSB full] on usb-0000:00:1d.0-1.6/input6

haata, 2016-02-22:
I just added support for dynamic USB power. It still has a few quirks with the Ergodox Interconnect, but this may help you out. I haven't pushed it to the configurator yet, but I will once I feel more comfortable with the changes.

https://github.com/kiibohd/controller/commit/46765e85c57e19b89754f4f9c26c8f18b35148e5

ec1oud, 2016-02-28:
I've been running the latest git (46765e8) for a couple days.  Now it doesn't reset so often, but instead one half or the other goes dead sometimes, while typing.  I have to re-plug it to continue typing, which is even more annoying.  Haven't narrowed it down any more than that so far.

haata, 2016-02-28:
Hmm, I may have run into another potential issue (related), that I had to
fix for the KType.

Try commenting out the code in this section ->
https://github.com/kiibohd/controller/blob/master/Scan/ISSILed/led_scan.c#L657

If the problem goes away, this is an easy fix. If not, I have some other
ideas.

ec1oud, 2016-03-05:
Yeah, commenting out the whole function body keeps the LEDs off, and the keyboard is more stable, but still resets occasionally.  If I run https://github.com/ibaranov-cp/Ergo_Infinity_Display it keeps working for several hours, but not indefinitely.

ec1oud, 2016-05-29:
I'm getting really sick of this... haven't been able to use the keyboard at all the last couple of months.

Nobody else has this problem, really?  Could it be a hardware defect then?  Or are people just collecting the keyboards and not using them?  Or nobody installed backlight LEDs, if that's what the problem is?

I got a USB volt/current/power meter on ebay, which says it takes a maximum of about 100mA for both halves, with all LEDs lit to the default low brightness that I have configured in my own KLL.  That should be fine for the computer. It makes no difference whether I plug it into a powered hub or into a motherboard USB port or even decouple both halves and plug them into the computer separately.  I've had different kernel versions at different times.  I've tried to keep up with your firmware patches.  The latest change "Adding auto-restart support whenever USB gets into an odd state" seems like a step backwards, maybe - although it's hard to tell because there has never been a stable firmware for this keyboard.  Many months after shipping already, now.  I tried today downloading and flashing firmware from the online configurator again.  Makes no difference.

Nothing helps.  Both halves are resetting too often to do much typing.  It has gotten worse since a few months ago, I think.  I have a zip file from the configurator from January; with that, I can do a bit of typing, but it still resets far too often, and the constant flashing of full-brightness backlight LEDs as it resets every few seconds is really annoying.

ec1oud, 2016-05-29:
Results do vary between machines though.  With the January 23 firmware from the online configurator, while plugged directly into my MacBook Pro (which is running Linux ATM), it's actually... not resetting, and I can type.  But with newer firmware, it will start resetting again, on that machine too.  Depending on which firmware, it seems like either the right half or the left half is more stable, if I plug them in separately.  So I tend to think this is probably not a hardware defect.

ec1oud, 2016-05-29:
Running https://github.com/fredizzimo/infinity_ergodox, the "led" branch, is much better.  So I probably won't bother switching back to this one, unless it becomes reliable and somehow more useful, some day.

svxsvx, 2016-10-12:
I'm not 100% sure yet but if this is the default firmware of the _Massdrop Infinity Ergodox_ then I have the same issue. It already annoys me before I even started to use the keyboard for real. So just assembled, plugged it in and already after a few minutes of inactivity: keyboard gets inactive. The display is still on but the keyboard doesn't accept any input. Also switching layer via L1 doesn't work anymore. Have to re-plug it to get it working again. Unusable.
By the way there is another guy on the Massdrop discussion site for the _Infinity Ergodox_ who has the same issue probably.

haata, 2016-10-12:
@svxsvx Try bringing up the debug console. Does it show any messages?

https://github.com/kiibohd/controller/wiki/Debugging

If you get disconnected from this as well, the OS is forceably resetting the USB device for some reason. Perhaps a bad USB port/cable, power issues, etc.

svxsvx, 2016-10-13:
Oh, thanks for the replay. Didn't expect a response that fast.
You use OSX, isn't it? Everything works as expected on a (slightly older) MacBook. By that I think I can put aside Hardware issues. Also issues with the port are kinda unlikely; I've tried it on 2 other notebooks with in total 5 different ports. All the same. More likely it is related to Linux or how Linux handles the issue probably. Both laptops run Linux, one also runs Windows XP.  On XP it works quite fine as well even though it reminds me regularly that the device supports higher speed and I should try port faster than USB 2.0. (The laptop has none. ;) )

Debug output I have to hand in later. Here on FreeBSD 10.2 I don't get the Virtual Serial Port:
`umodem0: <Virtual Serial Port - Status> on usbus1`
`umodem0: data interface 3, has no CM over data, has break`

svxsvx, 2016-10-14:
```
 Revision:      074134c47442fa28559598209e3d7d3f30326177
 Branch:        HEAD
 Tree Status:   Clean

 Repo Origin:   https://github.com/kiibohd/controller.git
 Commit Date:   2016-03-01 20:29:11 -0800
 Commit Author: Jacob Alexander <haata@kiibohd.com>
 Build Date:    2016-03-01 23:14:24 -0800
 Build OS:      Linux-4.4.1-2-ARCH
 Architecture:  arm
 Chip:          mk20dx256vlh7
 CPU:           cortex-m4
 Device:        Keyboard
 Modules:       Scan(MDErgo1) Macro(PartialMap) Output(pjrcUSB) Debug(full)
 Unique Id:     8CE1000067DA002230800128934E45
```
_(Default version shipped by Massdrop. No modification at all.)_

Yes, it also disconnects while debug console is open.
```
: INFO - Host has requested USB sleep/suspend state
INFO - USB Available Current Changed. Total AvailableWARNING - USB Transmit Timeout...
WARNING1;32mINFO - USB Available Current Changed. Total Available: 100 mA
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
Wmeout...
WARNING - USB Transmit Timeout...
(... last line repeated ~20 times ...)
```
The only difference I notice -- but maybe I mix it up: usually the L1-Key wasn't working as well anymore but this time it does. Anyway, any other input opt-out. (System: Gentoo Linux)

svxsvx, 2016-10-15:
Another update, since I've noticed the `outputDebug` command.
#### How To Reproduce
1. Connect the ErgoDox Infinity (only left half in my case)
2. `screen /dev/ttyACM0`
3. `:outputDebug`
4. Wait till it doesn't respond anymore (In my case: less than 3 minutes)
5. Type whatever in *screen*(8) with another keyboard to revitalize CLI as well as the ErgoDox

#### Debug Output
```
: INFO - Host has requested USB sleep/suspend state
INFO - USB Available Current Changed. Total AvailableWARNING - USB Transmit Timeout...
WARNING1;32mINFO - USB Available Current Changed. Total Available: 100 mA
DEBUG - NKRO USB: 00 000004000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 080000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000000000000 0000000000000000000000000000 00 000000000000
DEBUGmeout...
WARNING - USB Transmit Timeout...
WARNING - USB Transmit Timeout...
(.... last message repeated 18 times ...)
WARNING - USB Transmit Timeout...
DEBUG - NKRO USB: 00 000004000000 0000000000000000000000000000 00 000000000000
fsdDEBUG - NKRO USB: 00 000000000000 0000000000000000000000000000 00 000000000000
asDEBUG - NKRO USB: 00 080000000000 0000000000000000000000000000 00 000000000000
dDEBUG - NKRO USB: 00 000000000000 0000000000000000000000000000 00 000000000000
```

haata, 2016-10-15:
Hmm. It looks like the OS is telling the keyboard to go into low power mode.

USB power saving on Linux is usually disabled because it's extremely buggy. I need to setup a Linux machine that actually has this issue (Arch Linux does not). Most likely I'll need to modify some USB power saving code to handle the Linux corner case (Windows and Mac work fine).
At least for now, see if some of this helps -> http://unix.stackexchange.com/questions/91027/how-to-disable-usb-autosuspend-on-kernel-3-7-10-or-above

svxsvx, 2016-10-20:
> see if some of this helps -> http://unix.stackexchange.com/questions/91027/how-to-disable-usb-autosuspend-on-kernel-3-7-10-or-above
Set usbcore.autosuspend=-1 in *grub*:
```# cat /sys/module/usbcore/parameters/autosuspend
-1
```
but: nope.

For the reason I already found even more people with similar issues and one suggested just to update the keyboard layout I moved on. And while typing this, the keyboard is still alive -- usually would be opt-out already. So I will move on with the right-hand right after.

Just for further examination and reference I leave the new output of `version` of my left-side here (see above for old output for version):
```
: version
 Revision:      ba3c38bf844d0bea1681387ac8602e4ac1ba5f4c
 Branch:        master
 Tree Status:   Clean
 
 Repo Origin:   https://github.com/kiibohd/controller.git
 Commit Date:   2016-10-10 23:15:13 -0700
 Commit Author: GitHub <noreply@github.com>
 Build Date:    2016-10-19 03:37:22 +0000
 Build OS:      Linux-4.1.5-x86_64-linode61
 Architecture:  arm
 Chip:          mk20dx256vlh7
 CPU:           cortex-m4
 Device:        Keyboard
 Modules:       Scan(MDErgo1) Macro(PartialMap) Output(pjrcUSB) Debug(full)
 Unique Id:     8CE1000067DA00220030800128934E45
```
The difference is already noticeable right after entering the debug console. This escape sequence as the first output right at the start is gone: `^[[[11;32322mININNFOO^[[[00`

chadharrington, 2017-03-06:
I am seeing the same issue (left hand goes dead after a few minutes) on my brand-new Ergodox Infinity. I am using a Mac, not Linux. Is there a known fix for this on Macs?

haata, 2019-02-18:
I've fixed a lot of the USB suspend/resume code, please re-open if this is still an issue with the latest firmware.

closed 2019-02-18 by haata