kiibohd/controller #92: Adding support for addressable LED strips

nikchi, 2016-02-12:
This is a relatively recent development from QMK, the ability to use the VCC Data and GND pins on the microcontrollers to enable support for these types of LEDs

It would be really useful for the IC family of boards also.

https://github.com/jackhumbert/qmk_firmware/pull/106

haata, 2016-02-12:
Hmm, that shouldn't be that difficult. In theory it should be possible to use any free pin (not sure if it's possible to make it more efficient utilizing a hardware module though).
I had someone ping me more recently on helping out with something like this, perhaps it can be made to happen :D

smasher816, 2018-04-04:
Now that there is the PixelMap module and KLL support for aninamations this may be much more interesting to implement.